chrisjoyce911/esp32FOTA #1: There was no content in the response

wavresangam, 2019-03-27:
5:28:34.719 -> ets Jun  8 2016 00:22:57
15:28:34.719 -> 
15:28:34.719 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
15:28:34.719 -> configsip: 0, SPIWP:0xee
15:28:34.719 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
15:28:34.719 -> mode:DIO, clock div:1
15:28:34.719 -> load:0x3fff0018,len:4
15:28:34.719 -> load:0x3fff001c,len:928
15:28:34.752 -> ho 0 tail 12 room 4
15:28:34.752 -> load:0x40078000,len:9280
15:28:34.752 -> load:0x40080400,len:5848
15:28:34.752 -> entry 0x40080698
15:28:35.083 -> Connecting to ........Getting HTTP
15:28:39.171 -> http://192.168.10.110:8080/fota/fota.json
15:28:39.171 -> ------
15:28:39.669 -> Connecting to: 192.168.10.110
15:28:39.868 -> Fetching Bin: /fota/esp32-fota-http-2.bin
15:28:39.935 -> contentLength : 0, isValidContentType : 0
15:28:39.935 -> There was no content in the response
15:28:41.930 -> Getting HTTP
15:28:41.930 -> http://192.168.10.110:8080/fota/fota.json
15:28:41.930 -> ------
15:28:42.461 -> Connecting to: 192.168.10.110
15:28:42.595 -> Fetching Bin: /fota/esp32-fota-http-2.bin
15:28:42.661 -> contentLength : 0, isValidContentType : 0
15:28:42.661 -> There was no content in the response
15:28:44.652 -> Getting HTTP
15:28:44.652 -> http://192.168.10.110:8080/fota/fota.json
15:28:44.652 -> ------
15:28:44.718 -> Connecting to: 192.168.10.110
15:28:44.751 -> Fetching Bin: /fota/esp32-fota-http-2.bin
15:28:44.784 -> contentLength : 0, isValidContentType : 0
15:28:44.784 -> There was no content in the response
15:28:46.774 -> Getting HTTP
15:28:46.774 -> http://192.168.10.110:8080/fota/fota.json
15:28:46.774 -> ------
15:28:47.470 -> Connecting to: 192.168.10.110
15:28:47.571 -> Fetching Bin: /fota/esp32-fota-http-2.bin
15:28:47.636 -> contentLength : 0, isValidContentType : 0
15:28:47.669 -> There was no content in the response
15:28:49.660 -> Getting HTTP
15:28:49.660 -> http://192.168.10.110:8080/fota/fota.json
15:28:49.660 -> ------
15:28:49.892 -> Connecting to: 192.168.10.110
15:28:50.091 -> Fetching Bin: /fota/esp32-fota-http-2.bin
15:28:50.091 -> contentLength : 0, isValidContentType : 0
15:28:50.125 -> There was no content in the response
15:28:52.115 -> Getting HTTP
15:28:52.115 -> http://192.168.10.110:8080/fota/fota.json

wavresangam, 2019-03-27:
did not getting what happen to this, continuously getting this message

chrisjoyce911, 2019-03-27:
The logs are showing that http://192.168.10.110/fota/esp32-fota-http-2.bin has no content
try using `curl -i http://192.168.10.110/fota/esp32-fota-http-2.bin` and review what is being returned

wavresangam, 2019-03-27:
curl: (7) Failed to connect to 192.168.10.110 port 80: Connection refused

i am getting this output for curl -i http://192.168.10.110/fota/esp32-fota-http-2.bin  command

wavresangam, 2019-03-27:
but as i started local server on port 8080 and using curl command 
 curl -i http://192.168.10.110:8080/fota/esp32-fota-http-2.bin

and i am getting response

HTTP/1.1 200 OK
server: ecstatic-3.3.1
cache-control: max-age=3600
last-modified: Wed, 27 Mar 2019 08:52:23 GMT
etag: W/"3019655-782608-2019-03-27T08:52:23.453Z"
content-length: 782608
content-type: application/octet-stream; charset=utf-8
Date: Wed, 27 Mar 2019 11:59:27 GMT
Connection: keep-alive

Warning: Binary output can mess up your terminal. Use "--output -" to tell 
Warning: curl to output it to your terminal anyway, or consider "--output 
Warning: <FILE>" to save to a file.

chrisjoyce911, 2019-03-27:
There are two things to look at 

1) what did you set the  "port" to in your http://192.168.10.110:8080/fota/fota.json
`{
    "type": "esp32-fota-http",
    "version": 2,
    "host": "192.168.0.100",
    "port": 80,
    "bin": "/fota/esp32-fota-http-2.bin"
}`
2) add some debugging to esp32FOTA/src/esp32fota.cpp (line 36) to confirm the port is being set .

`  
    Serial.println("Connecting to: " + String(_host));
    Serial.println("Using port: " + String(_port));
    // Connect to Webserver
    if (client.connect(_host.c_str(), _port))
    {`

wavresangam, 2019-03-27:
{
    "type": "esp32-fota-http",
    "version": 2,
    "host": "192.168.10.110",
    "port": 8080,
    "bin": "/fota/esp32-fota-http-2.bin"
}

wavresangam, 2019-03-27:
[Wed Mar 27 2019 18:09:31 GMT+0530 (IST)] "GET /fota/fota.json" "ESP32HTTPClient"
[Wed Mar 27 2019 18:09:31 GMT+0530 (IST)] "GET /fota/esp32-fota-http-2.bin" "undefined"

and at server side i am getting above log 
first line when esp32  tried to get JSON file  as "ESP32HTTPClient" client
but in second line it is showing "undefined" client

wavresangam, 2019-03-27:
thank you for this fantastic library i completed FOTA, problem was related to server i solved that 
thank you once again...

closed 2019-03-27 by chrisjoyce911