chrisjoyce911/esp32FOTA #3: Error: control reaches end of non-void function

ipqate, 2019-04-03:
Hi,

ARDUINO IDE: 1.8.8
BOARD: ESP32
CODE: AS PER EXAMPLE PROVIDED

I'm trying to run the example but when compiling, I get this error message:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\PCIP\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\PCIP\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\PCIP\Documents\Arduino\libraries -fqbn=esp32:esp32:atelei:FlashFreq=80,PartitionScheme=default,UploadSpeed=921600,DebugLevel=none -ide-version=10808 -build-path C:\Users\PCIP\AppData\Local\Temp\arduino_build_997152 -warnings=all -build-cache C:\Users\PCIP\AppData\Local\Temp\arduino_cache_929599 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.0 -prefs=runtime.tools.esptool_py-2.6.0.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.0 -verbose C:\Users\PCIP\Documents\Arduino\libraries\esp32FOTA\examples\HTTP\http.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\PCIP\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\PCIP\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\PCIP\Documents\Arduino\libraries -fqbn=esp32:esp32:atelei:FlashFreq=80,PartitionScheme=default,UploadSpeed=921600,DebugLevel=none -ide-version=10808 -build-path C:\Users\PCIP\AppData\Local\Temp\arduino_build_997152 -warnings=all -build-cache C:\Users\PCIP\AppData\Local\Temp\arduino_cache_929599 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.0 -prefs=runtime.tools.esptool_py-2.6.0.path=C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.0 -verbose C:\Users\PCIP\Documents\Arduino\libraries\esp32FOTA\examples\HTTP\http.ino
Using board 'atelei' from platform in folder: C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1
Using core 'esp32' from platform in folder: C:\Users\PCIP\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.1
Opciones de compilación cambiadas, reconstruyendo todo
Detecting libraries used...
"C:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/config" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/app_trace" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/app_update" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/asio" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/bootloader_support" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/bt" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/coap" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/console" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/driver" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/esp-tls" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/esp32" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/esp_event" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/esp_http_client" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/esp_http_server" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/esp_https_ota" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/esp_https_server" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/ethernet" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/expat" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/fatfs" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/freemodbus" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/freertos" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/heap" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/idf_test" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/jsmn" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/json" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/libsodium" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/log" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/lwip" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/mbedtls" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/mdns" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/micro-ecc" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/mqtt" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/newlib" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/nghttp" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/nvs_flash" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/openssl" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/protobuf-c" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0 [... cut ...] l\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/smartconfig_ack" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/soc" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/spi_flash" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.1/tools/sdk/include/spiffs" "-IC:\\Users\\PCIP\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware

ipqate, 2019-04-03:
Solved! It is a problem when reaching the end of the function without giving back the result, in this case a boolean. Just add RETURN FALSE in the line no. 258.

Please check it:)

rljonesau, 2019-05-26:
Yes, I had the same issue, but no compiler warning.

In my instance execHTTPcheck does not return an explicit false when the "Is connected" test fails.
I was using AP only mode yet it said there was new firmware on the www server - um no not with AP only mode!

szundi, 2019-06-18:
this is fixed and merged now

closed 2019-06-18 by chrisjoyce911