chrisjoyce911/esp32FOTA #8: Other OTA options

iMina, 2019-09-15:
Great work ! Thank you !!
I already see a great infrastructure and very simplified interface. However, why not expand it ?
It would be really handy if the option to manually supply the fota.json instead of providing its URL.
Also the option to manually provide the OTA directly might be useful.
I am actually working on a project now that involves both, trying to find a workaround for them, and natively supporting them would be great !

chrisjoyce911, 2019-09-16:
You need some way to check if the current firmware is out of date, the firmware.json includes this information.

```
{
    "type": "esp32-fota-http",
    "version": 2,
    "host": "192.168.0.100",
    "port": 80,
    "bin": "/fota/esp32-fota-http-2.bin"
}
``

> manually provide the OTA directly

I'm guessing you are wanting a feature to do something like 
```
void loop()
{

  int newVersion = 2; 
  esp32FOTA.setHost =  "192.168.0.100";
  esp32FOTA.setPort =  80;
  esp32FOTA.setBin =  "/fota/esp32-fota-http-2.bin;

  bool updatedNeeded = esp32FOTA.execVersionCheck(newVersion);
  if (updatedNeeded)
  {
    esp32FOTA.execOTA();
  }

  delay(2000);
}
```

szundi, 2019-10-26:
Hey @chrisjoyce911, maybe we should write a helper function that includes your just posted code? This way the lib would have the feature.

chrisjoyce911, 2019-10-26:
I agree , it would be a simple improvement

szundi, 2019-10-30:
perfect, just take care about the typos. :)

chrisjoyce911, 2019-11-08:
I've just pushed up a solution [c8db6ef](https://github.com/chrisjoyce911/esp32FOTA/commits/c8db6ef) but have not tested it yet

chrisjoyce911, 2019-11-12:
Will include in next release

closed 2019-11-12 by chrisjoyce911

ZZKK000, 2022-11-08:
@chrisjoyce911 
Hello, If the esp32FOTA library can work with W5500 together with Ethernet2 library? or better with httpOTA?

tobozo, 2022-11-08:
hey @ZZKK000 it should work (already tested with enc28j60 and wt32-eth01) provided you maintain a variable with the connection state, and assign a getter to the FOTA object using `FOTA.setStatusChecker( fn )`

```cpp
bool eth_connected = false;

static bool EthernetConnected()
{
  return eth_connected;
}

// using WiFiEvent to handle Ethernet events :-)
void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case ARDUINO_EVENT_ETH_START:
      Serial.println("ETH Started");
      ETH.setHostname("esp32-ethernet");
      break;
    case ARDUINO_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case ARDUINO_EVENT_ETH_GOT_IP:
      Serial.printf("ETH MAC: %s, IPv4: %s%s, %dMbps\n", ETH.macAddress().c_str(), ETH.localIP().toString().c_str(), ETH.fullDuplex()?", FULL_DUPLEX":"", ETH.linkSpeed() );
      eth_connected = true;
      break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void setup()
{
  Serial.begin(115200);
  {
    auto cfg = FOTA.getConfig();
    cfg.name         = firmware_name;
    cfg.manifest_url = FOTA_URL;
    cfg.sem          = SemverClass( firmware_version_major, firmware_version_minor, firmware_version_patch );
    cfg.check_sig    = check_signature;
    cfg.unsafe       = disable_security;
    //cfg.root_ca      = MyRootCA;
    //cfg.pub_key      = MyRSAKey;
    FOTA.setConfig( cfg );
    FOTA.setStatusChecker( EthernetConnected );
  }
  WiFi.onEvent( WiFiEvent );
  ETH.begin();
}

```

ZZKK000, 2022-11-09:
@tobozo 
Thanks!

ZZKK000, 2022-11-09:
Hi, guys, there  are two kinds of OTA:
1. Firmware is stored in a file server, esp32 nodes get the firmware as http client actively.
2. Firmware is stored in a computer as web client and actively post the bin file to esp32 nodes which act as multiple web servers
if you have thousands esp32 nodes, which method is better?

tobozo, 2022-11-09:
@ZZKK000 although esp32FOTA has received features that seem industrial, it is still a hobby project and shouldn't be considered as production ready for any industrial use.

If you have thousands esp32 nodes I suggest you drop esp32FOTA and adopt a more industrial approach e.g. based on esp-idf and secureboot.

ZZKK000, 2022-11-18:
@tobozo 
Found this: 
https://github.com/maxgerhardt/pio-esp32-ethernet-ota