chrisjoyce911/esp32FOTA #10: Always non 200 response

iMina, 2019-09-17:
I am trying to test my OTA. I always get the same non 200 status code when I call execOTA()
```
Connecting to: spimesenselabs.com
Fetching Bin: /IoTik/v19.ino.bin
Got a non 200 status code from server. Exiting OTA Update.
contentLength : 0, isValidContentType : 0
There was no content in the response
```
And this is my JSON : 
```
{
    "type": "esp32-fota-http",
    "version": 20,
    "host": "spimesenselabs.com",
    "port": 80,
    "bin": "/IoTik/v19.ino.bin"
}
```
Any hints ?

chrisjoyce911, 2019-09-17:
I would test the location first using curl
```
$ curl --head http://spimesenselabs.com:80/IoTik/v19.ino.bin
```
then look for a **HTTP/1.1 200 OK** in the header

If this works then you need to confirm that your esp32  has network access to the server (including DNS)

iMina, 2019-09-17:
I tried curl and I got the following response
```
HTTP/1.1 200 OK
Server: openresty/1.13.6.2
Date: Tue, 17 Sep 2019 22:03:37 GMT
Content-Type: application/octet-stream
Content-Length: 1699584
Connection: keep-alive
Last-Modified: Mon, 16 Sep 2019 21:06:12 GMT
ETag: "67e5ca60-19ef00-592b1f93f38fa"
X-Webcom-Cache-Status: BYPASS
Accept-Ranges: bytes
```
Which in deed has **200 OK** in the header.
How can I confirm the esp32 access ?
Other libraries that rely on the network connection in deed work, for example a NTP time server I'm using.

chrisjoyce911, 2019-09-17:
You could try adding some debugging like
``` Serial.println("Got " + line); ``` somewhere before line 78 of esp32fota.cpp
https://github.com/chrisjoyce911/esp32FOTA/blob/752cc70056d3ef328eef445f866cdb8da835df5e/src/esp32fota.cpp#L78

This section checks for the 200 status

iMina, 2019-09-19:
I tried it and got `HTTP/1.1 400 Bad Request`
Any ideas ?

szundi, 2019-10-26:
@iMina have you found out what is the problem? Maybe try hosting it elsewhere for the sake of finding out wether it is server or client slide problem.

szundi, 2019-10-26:
Actually, I checked, I am hosting my firmwares on a google cloud static bucket and it works fine.

szundi, 2019-10-30:
As the bug report doesn't contain enough info and not reproducible on our end, I close this ticket. @iMina please feel free to open and post more info. Cheers.

closed 2019-10-30 by szundi

williamsantoso93, 2020-05-29:
Hi, I still have this problem.
My json file
```
{
    "type": "esp32-fota-http",
    "version": 3,
    "host": "http://leastric-device-ota.s3-ap-southeast-1.amazonaws.com",
    "port": 80,
    "bin": "/test/testfile.bin"
}
```

I tried curl too :
```curl  --head http://leastric-device-ota.s3-ap-southeast-1.amazonaws.com:80/test/testfile.bin```

the following response :
```
HTTP/1.1 200 OK
x-amz-id-2: z+rmECm3+0WRs9serOiuOklCMEpgk7TN2ooN4zPX5bjeG3799/sREvEP9GJrhoM+PTf/R2PEWZs=
x-amz-request-id: ASCH3HDQ8MEYCQEW
Date: Fri, 29 May 2020 08:14:09 GMT
Last-Modified: Fri, 29 May 2020 07:30:28 GMT
ETag: "59b6eb3fd0919051b7deae7141f8906f"
Accept-Ranges: bytes
Content-Type: application/macbinary
Content-Length: 874096
Server: AmazonS3
```

Debug result :
```
Connecting to: http://leastric-device-ota.s3-ap-southeast-1.amazonaws.com
Fetching Bin: /test/testfile.bin
Line = HTTP/1.1 400 Bad Request
Got a non 200 status code from server. Exiting OTA Update.
contentLength : 0, isValidContentType : 0
There was no content in the response
```

what should I do?

Thanks.

szundi, 2020-05-30:
@williamsantoso93 please remove "http://" from the hostname, I think this will do it. Cheers!

williamsantoso93, 2020-05-31:
Thanks for your info @szundi, I have solved that problem. 

Now I have a new problem. When I use the library with my current code, I cannot do the update

I got this result
```
Connecting to: leastric-device-ota.s3-ap-southeast-1.amazonaws.com
Fetching Bin: /test/FOTA_test.ino.doitESP32devkitV1.bin
Got application/macbinary payload.
Got 874096 bytes from server
contentLength : 874096, isValidContentType : 1
Begin OTA. This may take 2 - 5 mins to complete. Things might be quite for a while.. Patience!
E (95107) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (95140) task_wdt:  - async_tcp (CPU 1)
E (95140) task_wdt: Tasks currently running:
E (95140) task_wdt: CPU 0: ipc0
E (95140) task_wdt: CPU 1: loopTask
E (95140) task_wdt: Aborting.
abort() was called at PC 0x400e324b on core 0

Backtrace: 0x4008c820:0x3ffbe170 0x4008ca51:0x3ffbe190 0x400e324b:0x3ffbe1b0 0x40084b5d:0x3ffbe1d0 0x400828a9:0x3ffb9d60 0x40086e2e:0x3ffb9d80 0x40082f5b:0x3ffb9da0 0x40088f69:0x3ffb9dc0

Rebooting...
```

BTW I also use webserver.

Thanks.

GodayMax, 2022-07-19:
> 

meet the same problem......

szundi, 2022-07-20:
@GodayMax @williamsantoso93 please set up your watchdog to trigger after a longer time than it takes to make the firmware flashing. If you do not update the watchdog, it will reset your CPU before the flashing can finish. I would not simply disable though.