chrisjoyce911/esp32FOTA #15: Version info without json file from HTTP headers

szundi, 2019-11-16:
Hello,

I was thinking that maybe there is a way to remove the need for an additional json file that OTA uses to decide wether there is upgrade required. If the bin file HTTP headers would contain a version number, this could be enough to make a choice.

What do you think? Maybe I am biased as I am hosting my bin firmware files on Google Storage Buckets, and it is easy to set headers there.

András

chrisjoyce911, 2019-11-17:
I agree, my original thoughts where to use the json as its easy for anyone .

szundi, 2019-11-19:
And it is the correct MVP, as without the json solution, you cannot make this work on every environment. Json makes it possible to do this FOTA in cases when you cannot manipulate headers. Which is the case for the majority of our users who will use some kind of cloud solution, like Dropbox, etc.

When my time allows, I will make a PR for you to review.

chrisjoyce911, 2019-11-19:
Looking forward to reviewing it

closed 2022-04-25 by chrisjoyce911