chrisjoyce911/esp32FOTA #20: Header correction in execOTA function

sujeendran, 2020-02-06:
Hi, execOTA() function I noticed a typo that causes error when downloading the binary file. 
The part reading the headers is checking if the line starts with "Content-Type: " where it should be "Content-type: ". Making this change fixed my program.
```
if (line.startsWith("Content-type: "))
            {
                String contentType = getHeaderValue(line, "Content-type: ");
                Serial.println("Got " + contentType + " payload.");
                if (contentType == "application/octet-stream")
                {
                    isValidContentType = true;
                }
            }
```

chrisjoyce911, 2020-02-06:
I have made  the change could you please test 
https://github.com/chrisjoyce911/esp32FOTA/tree/issue20

sujeendran, 2020-02-06:
Tested. Works! Great quick fix!

DoubleChuang, 2020-02-06:
`message-header = field-name ":" [ field-value ]`

> Field_ names are case-insensitive.

Dharmik-Bhavani, 2020-04-13:
> Hi, execOTA() function I noticed a typo that causes error when downloading the binary file.
> The part reading the headers is checking if the line starts with "Content-Type: " where it should be [...]
I had an simmilar error but mine was fixed with having "Content-Type: " insted of "Content-type: ".
Not able to figure this out why it works sometimes with "Content-Type: " and sometimes with "Content-type: ".

So I made little change in the condition on line 94 in esp32fota.cpp file to 
if (line.startsWith("Content-type: ") || line.startsWith("Content-Type: "))

instead of
 if (line.startsWith("Content-type: "))

IoTThinks, 2020-06-06:
I also have the same issue.
Need to change to line.startsWith("Content-Type: ")
Or it is more reliable to compare ignoring the case.

szundi, 2020-07-08:
@Dharmik-Bhavani can you please submit a PR?

closed 2020-07-26 by chrisjoyce911