chrisjoyce911/esp32FOTA #26: Pull Request to support HTTPS

fbambusi, 2020-05-09:
Thank you for opening an issue on an esp32OTA Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **ESP32**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.10**

-I created the class secureEsp32FOTA, that can operate over https. I cloned the project and created a nwe branch, adding the class to the code as well as an example. I am not able to push the branch so as to create a Pull Request: could you please give me this possibility?

closed 2020-05-12 by chrisjoyce911

workpage2, 2020-10-31:
Can you put it here?

chrisjoyce911, 2020-11-01:
https://github.com/chrisjoyce911/esp32FOTA/blob/master/examples/HTTPS/HTTPS.ino

workpage2, 2020-11-01:
I used:

    secureEsp32FOTA secureEsp32FOTA("myavr", 1.0);
    secureEsp32FOTA._host="myavrupdate.ru"; //eg example.com
    secureEsp32FOTA._descriptionOfFirmwareURL="/update.json"; //e /my-fw-versions
And I get the error:

    Parsing failed

kloz8879, 2021-02-12:
> I used:
> 
> ```
> secureEsp32FOTA secureEsp32FOTA("myavr", 1.0);
> secureEsp32FOTA._host="myavrupdate.ru"; //eg example.com
> secureEsp32FOTA._descriptionOfFirmwareURL="/update.json"; //e /my-fw-versions
> ```
> 
> And I get the error:
> 
> ```
> Parsing failed
> ```

Make sure your JSON file is UTF-8 Encoded and the correct EOL characters are used (WIndows based encoding worked for me)