chrisjoyce911/esp32FOTA #28: Error compile HTTP example

williamsantoso93, 2020-05-28:
Hi Chris,
I have an error when I tried to compile HTTP example

error message :
Arduino: 1.8.12 (Mac OS X), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

/Users/williamsantoso/Documents/Arduino/libraries/esp32FOTA/src/esp32fota.cpp:271:8: error: prototype for 'String esp32FOTA::forceUpdate(String, int, String)' does not match any in class 'esp32FOTA'
 String esp32FOTA::forceUpdate(String firwmareHost, int firwmarePort, String firwmarePath)
        ^
In file included from /Users/williamsantoso/Documents/Arduino/libraries/esp32FOTA/src/esp32fota.cpp:8:0:
/Users/williamsantoso/Documents/Arduino/libraries/esp32FOTA/src/esp32fota.h:17:8: error: candidate is: void esp32FOTA::forceUpdate(String, int, String)
   void forceUpdate(String firwmareHost, int firwmarePort, String firwmarePath);
        ^
Multiple libraries were found for "WiFi.h"
 Used: /Users/williamsantoso/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Is there anything that's wrong?
Thanks

williamsantoso93, 2020-05-28:
Solved

closed 2020-05-28 by williamsantoso93

gustavoser78, 2020-06-04:
> Solved

Hello William , I have a very similar problem. How did you solve it?

C:\Users\gusta\Documents\Arduino\libraries\esp32FOTA\src\esp32fota.cpp:272:8: error: prototype for 'String esp32FOTA::forceUpdate(String, int, String)' does not match any in class 'esp32FOTA'

 String esp32FOTA::forceUpdate(String firwmareHost, int firwmarePort, String firwmarePath)

        ^

In file included from C:\Users\gusta\Documents\Arduino\libraries\esp32FOTA\src\esp32fota.cpp:8:0:

C:\Users\gusta\Documents\Arduino\libraries\esp32FOTA\src\esp32fota.h:17:8: error: candidate is: void esp32FOTA::forceUpdate(String, int, String)

   void forceUpdate(String firwmareHost, int firwmarePort, String firwmarePath);

        ^

Se encontraron varias bibliotecas para "WiFi.h"
Usado: C:\Users\gusta\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 No usado: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compilando para la tarjeta ESP32 Dev Module.

IoTThinks, 2020-06-06:
I have not put in the forceUpdate yet in my code.
Already hit the same problem.

Strange.

IoTThinks, 2020-06-06:
Solved.
Download this library as zip from Github and import to Arduino IDE works!

Installing the library via Arduino IDE will hit the compilation problem.
May be something is wrong during publishing the library to Arduino IDE?