chrisjoyce911/esp32FOTA #33: Content-Type validation error - not matching Type with type compare

wluque, 2020-06-09:
hello, I made the following correction for it to operate using iis hosting. change the T of type to uppercase.

#97   line.startsWith("Content-type: ")

IIS response : Content-Type  , T in upper case .

HTTP/1.1 200 OK
Content-Length: 874080
Content-Type: application/octet-stream
Last-Modified: Tue, 09 Jun 2020 15:13:54 GMT
Accept-Ranges: bytes
ETag: "065b996703ed61:0"
Server: Microsoft-IIS/10.0
X-Powered-By: ASP.NET
Date: Tue, 09 Jun 2020 15:40:00 GMT

very good job, works excellent !

best regards

IoTThinks, 2020-06-09:
For Apache, I changed like you, too.
Content-Type

mcedrdiego, 2020-06-11:
Thx mate. I had to do it for Apache too. 
PS:
    *Must Change line 97 and 99*

DoubleChuang, 2020-07-31:
Hi @wluque @IoTThinks @mcedrdiego,
I already fix it #20 and merged to master branch
please try it and I welcome your feedback.

closed 2021-02-13 by chrisjoyce911