chrisjoyce911/esp32FOTA #34: There was no content in the response

ramenia, 2020-07-08:
Hi,

I have a website so i wanted to try this library with that website. But i got "There was no content in the response" from uart terminal.

http://elifcalika.com/esp32/esp32.json
http://elifcalika.com/esp32/esp32.bin

`/**
   esp32 firmware OTA
   Purpose: Perform an OTA update from a bin located on a webserver (HTTP Only)
   Setup:
   Step 1 : Set your WiFi (ssid & password)
   Step 2 : set esp32fota()
   Upload:
   Step 1 : Menu > Sketch > Export Compiled Library. The bin file will be saved in the sketch folder (Menu > Sketch > Show Sketch folder)
   Step 2 : Upload it to your webserver
   Step 3 : Update your firmware JSON file ( see firwmareupdate )
*/

#include <esp32fota.h>
#include <WiFi.h>

// Change to your WiFi credentials
const char *ssid = "oguzhankalsin";
const char *password = "oguzhankalsin";

// esp32fota esp32fota("<Type of Firme for this device>", <this version>);
esp32FOTA esp32FOTA("esp32-fota-http", 1);

void setup()
{
  esp32FOTA.checkURL = "http://elifcalika.com/esp32/esp32.json";
  Serial.begin(115200);
  setup_wifi();
}

void setup_wifi()
{
  delay(10);
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println(WiFi.localIP());
}

void loop()
{
  delay(2000);
  esp32FOTA.forceUpdate("http://elifcalika.com", 80, "/esp32/esp32.bin");
}`

szundi, 2020-07-08:
@ramenia please remove the "http://" from the hostname.

ramenia, 2020-07-08:
Still same. 
```
contentLength: 1373584 isValidContentType: 0
There was no content in the response
```

http://elifcalika.com/esp32/esp32.json
http://elifcalika.com/esp32/esp32.bin
```
/**
   esp32 firmware OTA
   Purpose: Perform an OTA update from a bin located on a webserver (HTTP Only)
   Setup:
   Step 1 : Set your WiFi (ssid & password)
   Step 2 : set esp32fota()
   Upload:
   Step 1 : Menu > Sketch > Export Compiled Library. The bin file will be saved in the sketch folder (Menu > Sketch > Show Sketch folder)
   Step 2 : Upload it to your webserver
   Step 3 : Update your firmware JSON file ( see firwmareupdate )
*/

#include <esp32fota.h>
#include <WiFi.h>

// Change to your WiFi credentials
const char *ssid = "oguzhankalsin";
const char *password = "oguzhankalsin";

// esp32fota esp32fota("<Type of Firme for this device>", <this version>);
esp32FOTA esp32FOTA("esp32-fota-http", 1);

void setup()
{
  esp32FOTA.checkURL = "elifcalika.com/esp32/esp32.json";
  Serial.begin(115200);
  setup_wifi();
}

void setup_wifi()
{
  delay(10);
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println(WiFi.localIP());
}

void loop()
{
  delay(2000);
  esp32FOTA.forceUpdate("elifcalika.com", 80, "/esp32/esp32.bin");
}
```

mandria, 2020-07-21:
i had the same issue , it's just a typo in the library in the section for http request.
in the library it's written as "Content-type" instead of "Content-Type"

DoubleChuang, 2020-07-31:
Hi @mandria @ramenia,
I already fix it #20 and merged to master branch #35
please try it and I welcome your feedback.

ramenia, 2020-08-10:
Thank you it is solved.

closed 2020-08-10 by ramenia

hpsaturn, 2021-03-08:
Hi, I have the same issue, or maybe is similar:

I have the next output:

```
13:39:13.468 > Getting HTTP
13:39:13.468 > http://influxdb.canair.io:8080/releases/firmware.json
13:39:13.474 > ------
13:39:13.605 > FOTA: starting..
13:39:13.606 > Connecting to: influxdb.canair.io
13:39:13.680 > Fetching Bin: releases/canairio_TTGO_T7_rev736.bin
13:39:13.768 > Got a non 200 status code from server. Exiting OTA Update.
13:39:13.771 > contentLength : 0, isValidContentType : 0
13:39:13.777 > There was no content in the response
```
** My test files are:
http://influxdb.canair.io:8080/releases/canairio_TTGO_T7_rev735.bin
http://influxdb.canair.io:8080/releases/firmware.json

I'm using the library from Platformio registry and also via Github, but I don't understand what happen because with wget it works.

PR #102 (2022-09-30): 0.2.3
- flashZ support -> implements #100 implements #101
- gzip support -> fixes #25 
- bump version -> fixes #84 
- `::setStatusChecker( fn )` -> fixes #34 (need to test with #54)
- extended test suite (also fixes #25)