chrisjoyce911/esp32FOTA #38: Need clarification for example https

workpage2, 2020-10-05:
1. Where is the name of the firmware file for loading written?
2.What to write in the "description" line? Is it some kind of file? What's inside the file?

    secureEsp32FOTA secureEsp32FOTA("sensing-pot", 1);
    secureEsp32FOTA._host="firmwareHostWithoutPrependedHTTT.com"; //eg example.com
    secureEsp32FOTA._descriptionOfFirmwareURL="/path-to-firmware-description"; //e /my-fw-versions
    secureEsp32FOTA._certificate=test_root_ca;
    secureEsp32FOTA.clientForOta=clientForOta;

scubachristopher, 2020-10-05:
I struggled to deal with this myself.  I wanted to just force an update without version checking (no json) as I control what gets updated via MQTT and I wanted a straight path to just do the update.

I ended up using HTTPUpdate instead of the esp32FOTA:

```
#include <WiFiClientSecure.h>
#include <HTTPUpdate.h>

#define OTA_URL               "https://example.com/"

static WiFiClientSecure clientForOta;

static const char http_cert[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
[Put your cert here!]
-----END CERTIFICATE-----
)EOF";

...

void myExecOTA(void *firmwareFile) {
  char buff[80];

  strcpy(buff, OTA_URL);
  strcat(buff, (char *) firmwareFile);

  clientForOta.setCACert(http_cert);
  clientForOta.setTimeout(12000 / 1000);

  t_httpUpdate_return ret = httpUpdate.update(clientForOta, buff);
  switch (ret) {
    case HTTP_UPDATE_FAILED:
      Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());
      break;

    case HTTP_UPDATE_NO_UPDATES:
      Serial.println("HTTP_UPDATE_NO_UPDATES");
      break;

    case HTTP_UPDATE_OK:
      Serial.println("HTTP_UPDATE_OK");
      break;
  }
}

```

workpage2, 2020-10-06:
I put together the program from your example above. But it doesn't compile. Error in line: myExecOTA("ota");
Code:

    #include <WiFiClientSecure.h>
    #include <HTTPUpdate.h>

    #define OTA_URL               "https://example.com/"
    const char *ssid = "";
    const char *password = "";

    static WiFiClientSecure clientForOta;

    const char* ca = \ 
    "-----BEGIN CERTIFICATE-----\n" \
 
    "-----END CERTIFICATE-----\n";

    void setup_wifi()
    {
    delay(10);
    Serial.print("Connecting to ");
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED)
    {
      delay(500);
      Serial.print(".");
    }
  
    }

    void setup()
    {
      Serial.begin(115200);
      setup_wifi();
      Serial.println("update");
      myExecOTA("ota");
    }

    void myExecOTA(void *firmwareFile) {
      char buff[80];

      strcpy(buff, OTA_URL);
      strcat(buff, (char *) firmwareFile);

      clientForOta.setCACert(ca);
      clientForOta.setTimeout(12000 / 1000);

      t_httpUpdate_return ret = httpUpdate.update(clientForOta, buff);
      switch (ret) {
        case HTTP_UPDATE_FAILED:
          Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", httpUpdate.getLastError(), 
    httpUpdate.getLastErrorString().c_str());
          break;

        case HTTP_UPDATE_NO_UPDATES:
          Serial.println("HTTP_UPDATE_NO_UPDATES");
          break;

        case HTTP_UPDATE_OK:
          Serial.println("HTTP_UPDATE_OK");
          break;
      }
    }

    void loop()
    {
  
     delay(2000);
     Serial.print("#");
    }

scubachristopher, 2020-10-06:
Two issues:
1). You need to declare myExecOTA before calling, so put
`
void myExecOTA(void *firmwareFile);
`
above setup_wifi, and

2).  You need to cast "OTA" as a void *:

`
myExecOTA((void *) "ota");
`

This compiles under Arduino for me:

```
#include <WiFiClientSecure.h>
#include <HTTPUpdate.h>

#define OTA_URL               "https://example.com/"
const char *ssid = "";
const char *password = "";

static WiFiClientSecure clientForOta;

const char* ca = \ 
"-----BEGIN CERTIFICATE-----\n" \

"-----END CERTIFICATE-----\n";

void myExecOTA(void *firmwareFile);

void setup_wifi()
{
delay(10);
Serial.print("Connecting to ");
WiFi.begin(ssid, password);
while (WiFi.status() != WL_CONNECTED)
{
  delay(500);
  Serial.print(".");
}

}

void setup()
{
  Serial.begin(115200);
  setup_wifi();
  Serial.println("update");
  myExecOTA((void *) "ota");
}

void myExecOTA(void *firmwareFile) {
  char buff[80];

  strcpy(buff, OTA_URL);
  strcat(buff, (char *) firmwareFile);

  clientForOta.setCACert(ca);
  clientForOta.setTimeout(12000 / 1000);

  t_httpUpdate_return ret = httpUpdate.update(clientForOta, buff);
  switch (ret) {
    case HTTP_UPDATE_FAILED:
      Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", httpUpdate.getLastError(), 
httpUpdate.getLastErrorString().c_str());
      break;

    case HTTP_UPDATE_NO_UPDATES:
      Serial.println("HTTP_UPDATE_NO_UPDATES");
      break;

    case HTTP_UPDATE_OK:
      Serial.println("HTTP_UPDATE_OK");
      break;
  }
}

void loop()
{

 delay(2000);
 Serial.print("#");
}

```

workpage2, 2020-10-31:
Have you tried this example on a real https server? I keep getting the error

    HTTP_UPDATE_FAILED Error (-1): HTTP error: connection refused

Adminius, 2020-11-20:
Hi @workpage2 
I've updated a example for https. Can you verify if it's understandable and working for you?

closed 2021-02-13 by chrisjoyce911