chrisjoyce911/esp32FOTA #45: Resolve DNS of firmware update server

garageeks, 2021-02-12:
Dear all,
The Library is straightforward to use and works as expected.

However there is a limitation: while the check update URL can be a normal HTTP url to a web server, which is resolved by DNS, the server where the firmware is hosted must be an IP address; with a hostname it doesn't work because WiFiClient is used, instead of HTTPClient. This means upgrade is only possible from a local server or a web server with an exclusive IP address. 
Web site hosting wouldn't work as they share the same IP address for several websites.

I suggest to use HTTPClient, which has built in DNS instead of WiFiClient, to download the firmware.

IoTThinks, 2021-02-13:
I remember the library is doing well with DNS.

Check if your board is able to get DNS from DHCP. Or you can set DNS when using static IP.

chrisjoyce911, 2021-02-13:
@IoTThinks  I think you are correct , a quick review of https://www.arduino.cc/en/Reference/WiFiClientConnect shows it should.

@garageeks , can you please text that your wifi is supply a valid DNS and access

garageeks, 2021-02-13:
My apologies guys, you are right. My mistake was in the server JSON, I was passing as host an address with the http URL prefix. The correct answer is the second level domain, without the http prefix. This is required in the checkURL.

Passing wrong URL: 
```
http://redacted.com/fota_firmware.php?id=4675224752999
------
Connecting to: http://redacted.com
[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for http://redacted.com
Connection to http://redacted.com failed. Please check your setup
contentLength : 0, isValidContentType : 0
There was no content in the response
```

With the correct host name:

```
Getting HTTP
http://redacted.com/fota_firmware.php?id=4675224752999
------
Connecting to: redacted.com
Fetching Bin: /fota/firmware.bin
Got 956832 bytes from server
Got application/octet-stream payload.
contentLength : 956832, isValidContentType : 1
Begin FW OTA...
Written : 956832 successfully
OTA done!
Update successfully completed. Rebooting.
```

closed 2021-02-13 by garageeks