chrisjoyce911/esp32FOTA #49: HTTPS example returning "X509 - Certificate verification failed" error

eyecee, 2021-03-02:
Hi,

I'm trying to use the HTTPS example to update my ESP32 using a self-signed cert. I can successfully read the JSON file via HTTPS using the HTTPClient object and defining the cert in the http.begin(FWhost, FWport, FWfolder, CA_CERT) function so I know the ESP32 can access the HTTPS site, read and parse the JSON and complete the IF statement to trigger the executeOTA function.

However, I then encounter the following error:

``` 
02:54:20.655 -> [V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 264256
02:54:20.655 -> [V][ssl_client.cpp:58] start_ssl_client(): Starting socket
02:54:20.655 -> [V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
02:54:20.655 -> [V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
02:54:20.655 -> [V][ssl_client.cpp:115] start_ssl_client(): Loading CA cert
02:54:20.655 -> [V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
02:54:20.655 -> [V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
02:54:20.701 -> [E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
02:54:20.701 -> [E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -9984
02:54:20.701 -> [V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
```

Any advice on how to resolve this issue as I think it's tripping up on the following in the executeOTA function:
```
if(client.connect("server",443) {
``` 
Thanks in advance!

dexion, 2021-07-21:
try to set `client.setInsecure()` for ignoring all errors with certs

chrisjoyce911, 2022-04-25:
YOu can also test using a upcoming patch https://github.com/chrisjoyce911/esp32FOTA/pull/79

closed 2022-04-25 by chrisjoyce911