chrisjoyce911/esp32FOTA #54: TinyGSM and FOTA

grichfitz, 2021-03-11:
Hay peeps,

OK so this isnt really an issue, this FOTA works well. Thanks chris.

But can it work via GSM in particular TinyGSMclient?

I tried, but obviously it fails.

Just wondered if anyone had any input.

Thanks all.
grichfitz

hilsonp, 2021-03-17:
Hi,

I would be very interested too.

I had not time to test yet but maybe changing the following:
// OTA Logic
void esp32FOTA::execOTA()
{
    WiFiClient client;

By this would make the trick:
// OTA Logic
void esp32FOTA::execOTA()
{
    TinyGsmClient client(modem);

woodlist, 2021-04-23:
The usage with TinyGSM I think, should look like this (but it is impossible for given lib, for given moment): 
#include <TinyGSM.h>
#include someSSL.h
#include esp32fota.h

`TinyGSM modem(Serial1);
TinyGSMClient baseclient(modem);
someSSLClient sslclient(baseclient);
esp32Fota (sslclient);
`

IoTThinks, 2021-06-11:
A few more places are using WiFi object too. If use GSM, I expect to have more changes.
https://github.com/chrisjoyce911/esp32FOTA/blob/master/src/esp32fota.cpp#L216

IoTThinks, 2021-06-12:
After one hour, I think it is quite difficult to integrate Tiny GSM into this library.
The current library uses WiFiClient and HttpClient which are not friendly with Tiny GSM stack.

IoTThinks, 2021-06-13:
I forked this library into esp32FOTAGSM.
Able to...compile it now. 

Not run yet. Still in progress.

IoTThinks, 2021-06-15:
Working with TinyGSM. https://github.com/IoTThinks/esp32FOTAGSM
Changed quite a few places
- Fixed peek() in this library.
- Replaced WiFiClient by TinyGSM.
- Replaced HTTPClient by ArduinoHttpClient.
- Broke checkURL into checkHOST, checkPORT, checkRESOURCE
- Support HTTP only.

chupachupbum, 2022-08-24:
I made a fork that now working with TinyGsm https, by changing WifiClient with SSLClient.
https://github.com/chupachupbum/esp32FOTA

tobozo, 2024-08-19:
closing this as *maybe implemented* (still needs testing), feel free to add followup if something new comes up

closed 2024-08-19 by tobozo