chrisjoyce911/esp32FOTA #62: library in PlatformIO registry fails (we need re publish it?)

hpsaturn, 2021-11-20:
## Description

When I change the declaration of esp32FOTA library in my PlatformIO ini to last version, it fails. When I change it to the Github target, it compile ok. I think that is only because the version on PlatformIO registry is old, February, and produce the next error:

## Output

```cpp
lib/canairioota/src/OTAHandler.cpp: In member function 'void OTAHandler::checkRemoteOTA(bool)':
lib/canairioota/src/OTAHandler.cpp:60:40: error: 'class esp32FOTA' has no member named 'getPayloadVersion'
             _onUpdateMsgCb(String(fota.getPayloadVersion()).c_str());
                                        ^
Compiling .pio/build/TTGO_TDISPLAY/FrameworkArduino/MD5Builder.cpp.o
*** [.pio/build/TTGO_TDISPLAY/lib043/canairioota/OTAHandler.cpp.o] Error 1
```

## Config fail

```python
lib_deps = 
	chrisjoyce911/esp32FOTA @ ^0.1.4
```

## Workaround

```python
lib_deps = 
	; chrisjoyce911/esp32FOTA @ ^0.1.4
	https://github.com/chrisjoyce911/esp32FOTA.git
```

## Fix

We need a new tag a new version?

chrisjoyce911, 2021-11-21:
I don't use PlatformIO , can you please do a pull request

hpsaturn, 2021-11-21:
it is impossible because it is a problem in the deployment, maybe is a problem with PlatformIO registry, we was need make a new tag with a new revision number, but I can see that you already change the library.json manifiest and did a new tag, maybe with that, after some hours the problem will be fixed automatic.

Maybe in the future like good practice, you can make this changes more fast, publishing the library with [CLI of PlatformIO](https://docs.platformio.org/en/latest/librarymanager/creating.html)

On other hand, PlatformIO it is important, a lot of Arduino users [for example](https://nerdytechy.com/platformio-vscode-vs-arduino-ide/), using it. Espressif and many brands too. I have around the six yeas using it, and the community growing up too much.

hpsaturn, 2021-11-24:
it is working now again. Close.

closed 2021-11-24 by hpsaturn