chrisjoyce911/esp32FOTA #63: Esp32 will compile two bin files, one is app and the other is bootstrap. How to upgrade the two files via OTA?

qutaojiao, 2021-11-22:
Esp32 will compile two bin files, one is app and the other is bootstrap. How to upgrade the two files via OTA?

[image]

hpsaturn, 2021-11-24:
you only need this extra binary files for the base firmware, after the first flash you only need the main binary. In your example you only need to upload `ESP32-CAM_Motion.ino.partitions.bin` to your server, partitions not.

qutaojiao, 2021-11-25:
> 您只需要基本固件的额外二进制文件，在第一次闪存后，您只需要主二进制文件。在你的例子中，你只需要上传`ESP32-CAM_Motion.ino.partitions.bin`到你的服务器，分区不需要。

I see. Thank you very much!

marcelstoer, 2022-03-22:
@hpsaturn isn't your answer inconsistent?

> you only need the main binary. 

Correct.

> In your example you only need to upload `ESP32-CAM_Motion.ino.partitions.bin` to your server, partitions not.

"partitions not" is correct but the file you mentioned actually _is_ the partitions binary, isn't it. Didn't you mean to write `ESP32-CAM_Motion.ino.bin`? People working with PlatformIO would find a `firmware.bin` in the `<project>/.pio/build/<env>` folder (that also contains a `partitions.bin`).

hpsaturn, 2022-03-23:
yes, when you use Platformio you only needs loading in the server the firmware.bin, and no more, this file contains all, the firmware and the partitions info. I don't know how is it called with Arduino IDE

closed 2022-04-25 by chrisjoyce911