chrisjoyce911/esp32FOTA #71: No CRC or fall back

daverathbone, 2022-01-23:
Thank you for opening an issue on an esp32OTA Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

thorrak, 2022-01-26:
Can you link one of those examples you are referencing?

daverathbone, 2022-01-26:
Within Espressif's examples and API's see [https://github.com/espressif/esp-idf/blob/master/examples/system/ota/simple_ota_example/main/simple_ota_example.c](url) 

I first tested with your code only as I need other Arduino framework items for  ease of development within Platformio.
(Thank you)

As I outlined it would be handy if the Serial print or log errors came back as a call back, to the users main loop.
Allowing for user to error handling stuff, rather than modifying your library. 

I also found that I did not need certificate based handshakes for my HTTPS server as its encrypted any way. (_HTTPS ssl_)   
I just send  a username and password, in the HTTPs  'get' header which an Apache server uses to allow the JSON file access.
This is well secure enough, as the JSON reply has the real url for the firmware bin in it, which is well hidden from any 'man  in the middle' risks or server sniffing Bots.  

The standard trick is to send servers massive urls in order to break buffers., home grown node.js servers will fall over  other servers are known to spill the beans.  In the main it can just give you denial of service risks.

I've modified your library to use a local certificate or not on  a setup call.
Just remember that many self signed SSL certs,  will run out quicker and your embedded device will be lost at sea.

Note: The Arduino library to your code downloads from Platformio and gets the older version that does not have http or https switch . Which is up-to-date on this GIT repo.

PR #95 (2022-09-19): v0.2.1
- New callback setters (need tuning/feedback) fixes #87 fixes #89 :
  * `setUpdateBeginFailCb`
  * `setUpdateEndCb`
  * `setUpdateCheckFailCb`
  * `setUpdateFinishedCb`
- Arduino Library Manager compliance (now uses `#include <esp32FOTA.hpp>`) fixes #85 fixes #93 
- New constructor `esp32fota( FOTAConfig_t )` for init with predefined config, fixes #83 
- New empty constructor to use with `getConfig()` / `setConfig(FOTAConfig_t)` for late init
- Improved signature check, fixes #88 fixes #71 
- Wrapped C `semver_t` in C++ `SemverClass` 
- Added signed example to test suite
- `execOTA()` n [... cut]
closed 2022-09-19 by tobozo