chrisjoyce911/esp32FOTA #74: File reads are only compatible with SPIFFS

thorrak, 2022-01-23:
As written, this library assumes that users are using SPIFFS. An example of this is:

https://github.com/chrisjoyce911/esp32FOTA/blob/6648c9d5d1e9f9188cd30915588875eba88caec0/src/esp32fota.cpp#L52

It would be good to be able to select between SPIFFS and LittleFS for users that might have switched over.

Darren777, 2022-04-25:
I'm curious as to why SPIFFS was chosen as the filesystem for this project ? From my understanding LIttleFS is an upgrade to SPIFFS and considering how good & relevant this library is, I thought that would be using it. If not I'd love to know why not.

chrisjoyce911, 2022-04-25:
I would like to have support for both , as SPIFFS is not the primary function of this library it should not have a strong option forcing the use on or over the other. 

You could help out with come testing of  https://github.com/chrisjoyce911/esp32FOTA/pull/79 and assist in resolving any conflicts and or errors

tobozo, 2022-09-30:
bump

multiple filesystem is implemented in the last relase (see the anyFS example), closing this as solved

closed 2022-09-30 by tobozo