chrisjoyce911/esp32FOTA #78: X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected

rhamaa, 2022-01-25:
i got an error when i want to use HTTPS with root_ca.pem

Here is the log
```log
[V][HTTPClient.cpp:245] beginInternal(): url: https://192.168.18.192/x
[D][HTTPClient.cpp:293] beginInternal(): protocol: https, host: 192.168.18.192 port: 443 url: /x
[D][HTTPClient.cpp:579] sendRequest(): request type: 'GET' redirCount: 0

[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 267404
[V][ssl_client.cpp:65] start_ssl_client(): Starting socket
[V][ssl_client.cpp:104] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:113] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:129] start_ssl_client(): Loading CA cert
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():138]: (-8576) X509 - The CRT/CRL/CSR format is invalid, e.g. different type expected
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -8576
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
[D][HTTPClient.cpp:1118] connect(): failed connect to 192.168.18.192:443
[W][HTTPClient.cpp:1417] returnError(): error(-1): connection refused
[E][esp32fota.cpp:444] execHTTPcheck(): Error on HTTP request
[D][HTTPClient.cpp:400] disconnect(): tcp is closed

[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
```

daverathbone, 2022-01-26:
I also had this error, no matter what cert used.. Fix was not using a SSL cert  on ESP32 end but embedding user and password in Get  header.  Don't forget your encrypted anyway with HTTPS get. Then setting your server to only except file transfer on getting the embedded header values . It also stops your ESP32 having an out of date SSL cert and never updating via OTA.

chrisjoyce911, 2022-04-25:
You might like to test an upcoming patch https://github.com/chrisjoyce911/esp32FOTA/pull/79

closed 2022-04-25 by chrisjoyce911