chrisjoyce911/esp32FOTA #82: Ota never start connected via LTE

tradexsrl, 2022-03-01:
Hi , just discover your library , installed it and created a simple sketch to test it but it never do the execOTA() because execHTTPcheck() everytime return 0
i'm using a lilygo t-pcie with a simcom 7600 lte module , also with forceUpdate() it never do the update.
i have tested the server via a smartphone and i was able to get to the json file
board is connected to internet via Vodafone LTE metwork and i have an ubuntu server with ngnix setup to host json file and bin firmware file
can you help me debug to find where is the problem ?

NPluta, 2022-03-01:
Try your Webserver with postman and double  check if the response is sent as json.
Check if the files are accessable. 
Check if the filepath and name are correct. Is there a capital letter etc.
Can you check the access log on your server. Will the json file be requested?
Can you attach the .ino code?

tradexsrl, 2022-03-01:
files are accessible because i have checked this via smartphone web browser , i just put the url of the json file and i see it , the same for bin file , web browser askme to download it
[TinyGSM_Allfunctions.txt](https://github.com/chrisjoyce911/esp32FOTA/files/8162121/TinyGSM_Allfunctions.txt)

tradexsrl, 2022-03-01:
this is the postman check response:
[image]

NPluta, 2022-03-01:
Try to Change the Port to 80 for the file.
As far as i know the file cannot be downloaded with a port.

NPluta, 2022-03-01:
Good Point. I'm just another user try to help.
I remember having a problem with it. Try these steps:
Copy esp32fota.cpp and esp32fota.h into your ino directory.
comment lines 290 through 298 in esp32fota.cpp and dont use the forceupdate function. 
comment line 492
Did you give it around 5 minutes time to do the update? Can take a long time if connection is slow. 
Does it start the update and failed to download. Can you narrow down where it fails exactly?

tradexsrl, 2022-03-01:
thank's. do you mean comment lines of the files copied into ino dir or in the files of esp32fota in the arduino library ?
the lines number you ask to comment in my case are these:
[image]
[image]
i don't think are what you mean

tradexsrl, 2022-03-01:
also just copying the two file into the ino dir the compile process fails...

tradexsrl, 2022-03-01:
found a thread in this repo that say this library doesn't work because it use WIFI....a user made a fork so i will give it a try ....

NPluta, 2022-03-01:
I tried your server update:
16:14:33.791 -> Getting HTTP
16:14:33.791 -> ------
16:14:34.090 -> Update needed: 
16:14:34.090 -> [15154] 1
16:14:44.064 -> [25154] Ota procedure started
16:14:44.102 -> Connecting to: ota.xxx.com:80
16:14:44.139 -> Fetching Bin: /smart-trap-esp32/esp32-fota-http-2.bin
16:14:44.177 -> Got a non 200 status code from server. Exiting OTA Update.
16:14:44.177 -> contentLength : 0, isValidContentType : 0
16:14:44.177 -> There was no content in the response

NPluta, 2022-03-01:
Sorry for the comment thing. You should comment the force update function. 

// Force a firmware update regardless on current version
void esp32FOTA::forceUpdate(String firmwareURL, boolean validate )
{
    _firmwareUrl = firmwareURL;
    _check_sig = validate;
    execOTA();
}

tradexsrl, 2022-03-01:
you cannot test because ota.xxx.com is a fake url for privacy reason....

NPluta, 2022-03-01:
I got the right url.

NPluta, 2022-03-01:
Can you change the port to 81 or 80 on both.
Now its just for the bin file.

tradexsrl, 2022-03-01:
you are able to get to "onnecting to:" because you are using a wifi connection....try with a gsm connection

tobozo, 2022-12-06:
Here's an [esp32FOTA example](https://github.com/tobozo/ESP32-ENC28J60/blob/main/examples/FOTA/FOTA.ino) demontrating esp32FOTA using Ethernet instead of WiFi.

The Ethernet driver integration uses lwip stack from esp32 core, and makes use of WiFiEvent_t and other shared functionalities from WiFi.h, this is now the official way of doing network with ESP32 in Arduino IDE.

https://github.com/espressif/arduino-esp32/tree/master/libraries/Ethernet

closing this as stalled