chrisjoyce911/esp32FOTA #83: Not clear instruction on REAME: check SPIFFS with root_ca.pem (https) and rsa_key.pem (signature check)

milkywade, 2022-04-10:
This is a question / request for clarification

I am trying to make this example work: https://github.com/chrisjoyce911/esp32FOTA/blob/master/examples/HTTP/HTTPS.ino

In the example, it is not clear how to perform this step:
  Step 3 : Provide SPIFFS filesystem with root_ca.pem of your webserver

I also checked the README.md - same cryptic instruction "For https or signature check: SPIFFS with root_ca.pem (https) and rsa_key.pem (signature check)"

How does actually one do that ?

Can the certificate check not be managed by a constant in the sketch as described in this article: https://www.lab4iot.com/2021/02/21/esp32-secure-firmware-update-over-the-air-ota/?unapproved=660&moderation-hash=f37b98b9077b522d86cd237ae1a442dc#comment-660

char *siteCertificate = 
"-----BEGIN CERTIFICATE-----\n"
...

Many thanks in advance

chrisjoyce911, 2022-04-25:
We are close to some changes around this , you may wish to try the changes,

https://github.com/chrisjoyce911/esp32FOTA/pull/79

See example

https://github.com/thorrak/esp32FOTA/blob/c60cb7e4d76e22aae1003b52f46a26dfd8db70e8/examples/HTTP/HTTPS_without_root_cert.ino

milkywade, 2022-04-25:
tried to compile w the updated example, using v0.1.6, failed with "candidate expects 4 arguments, 5 provided", guess will have to wait until the new version has been committed

PR #95 (2022-09-19): v0.2.1
- New callback setters (need tuning/feedback) fixes #87 fixes #89 :
  * `setUpdateBeginFailCb`
  * `setUpdateEndCb`
  * `setUpdateCheckFailCb`
  * `setUpdateFinishedCb`
- Arduino Library Manager compliance (now uses `#include <esp32FOTA.hpp>`) fixes #85 fixes #93 
- New constructor `esp32fota( FOTAConfig_t )` for init with predefined config, fixes #83 
- New empty constructor to use with `getConfig()` / `setConfig(FOTAConfig_t)` for late init
- Improved signature check, fixes #88 fixes #71 
- Wrapped C `semver_t` in C++ `SemverClass` 
- Added signed example to test suite
- `execOTA()` n [... cut]
closed 2022-09-19 by tobozo