chrisjoyce911/esp32FOTA #84: HTTP/1.1 301 Moved Permanently

DonatelloX, 2022-04-21:
- Arduino board:  ESP32 Dev Module
- Arduino IDE 1.8.19
- ESP32 Core 1.0.6

Hello everyone,
congratulations for the useful library.
I am testing on different hostings with HTTP example without root certificate. All hosting are accessible with HTTPS. With the example code I am able to upgrade without problems on some hosting.
While on others it doesn't work with an `HTTP/1.1 301 Moved Permanently`, as detailed below.
Please can I get some help?
Thanks you

```
[I][esp32fota.cpp:214] execHTTPcheck(): Getting HTTP: https://****.eu/****/program.json
[I][esp32fota.cpp:215] execHTTPcheck(): ------
[V][HTTPClient.cpp:245] beginInternal(): url: https://****.eu/****/program.json
[W][HTTPClient.cpp:257] beginInternal(): unexpected protocol: https, expected http
[V][HTTPClient.cpp:245] beginInternal(): url: https://****.eu/****/program.json
[D][HTTPClient.cpp:293] beginInternal(): protocol: https, host: ****.eu port: 443 url: /****/program.json
[D][HTTPClient.cpp:579] sendRequest(): request type: 'GET' redirCount: 0

[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 274168
[V][ssl_client.cpp:65] start_ssl_client(): Starting socket
[V][ssl_client.cpp:104] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:113] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[V][ssl_client.cpp:197] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:212] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:233] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:242] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:257] start_ssl_client(): Free internal heap after TLS 230320
[D][HTTPClient.cpp:1125] connect():  connected to ****.eu:443
[V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 156 bytes...
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'HTTP/1.1 301 Moved Permanently'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Server: aruba-proxy'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Date: Thu, 21 Apr 2022 08:20:17 GMT'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Content-Type: text/html'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Transfer-Encoding: chunked'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Connection: keep-alive'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'Location: https://www.****.eu/****/program.json'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: 'X-ServerName: ipvsproxy135.ad.aruba.it'
[V][HTTPClient.cpp:1216] handleHeaderResponse(): RX: ''
[D][HTTPClient.cpp:1257] handleHeaderResponse(): code: 301
[D][HTTPClient.cpp:1264] handleHeaderResponse(): Transfer-Encoding: chunked
[D][HTTPClient.cpp:603] sendRequest(): sendRequest code=301

[E][esp32fota.cpp:260] execHTTPcheck(): Error on HTTP request
[D][HTTPClient.cpp:378] disconnect(): still data in buffer (179), clean up.

[D][HTTPClient.cpp:385] disconnect(): tcp keep open for reuse
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.
```

DonatelloX, 2022-04-21:
I changed the port in JSON information file from 80 to 443, then I downloaded the latest version 0.1.6 which is not automatically visible in Arduino. With these steps it seems to work.
Please, can you insert the new version in Arduino? Thank you

chrisjoyce911, 2022-04-25:
Almost ready for a new release , you could help test 
https://github.com/chrisjoyce911/esp32FOTA/pull/79

PR #102 (2022-09-30): 0.2.3
- flashZ support -> implements #100 implements #101
- gzip support -> fixes #25 
- bump version -> fixes #84 
- `::setStatusChecker( fn )` -> fixes #34 (need to test with #54)
- extended test suite (also fixes #25)
closed 2022-09-30 by tobozo