chrisjoyce911/esp32FOTA #87: Reporting status during the update process

nonsintetic, 2022-05-18:
It would be very useful to have a way of getting some updates what's going on during an update. It would let you, for example, publish some messages or errors to MQTT or BLE while the update is going on.

Think a simple way to do this would be to have a callback function that gets passed on initialization that then gets called whenever something happens during the update process. 

Things that could be useful to track in my opinion: errors, download status for firmware file, start of the actual update process once the file is downloaded, result (success/fail).

tobozo, 2022-09-16:
Hi,

Not exactly what you're looking for but you might be interested by the new `esp32fota.setProgressCb( callback )` method added in [0.2.0](https://github.com/chrisjoyce911/esp32FOTA/releases/tag/0.2.0).

Based on your input, here are some ideas for callbacks which could send MQTT messages without disturbing the update process:

```C++
// when Update.begin() returned false
onUpdateBeginFail( int partition ); 

// after Update.end() and before validate_sig()
onUpdateEnd( int partition ); 

// validate_sig() error handling, mixed situations
onUpdateCheckFail( int partition, int error_code ); 

// update successful
onUpdateFinished( int partition, bool restart_after ); 

// Note: int partition; can be U_SPIFFS or U_FLASH

```

Please let me know if you're still looking for such features and I'll prioritize this. 
I'm on my way to write a 0.2.1 version of this library (adding gzip support) and this is totally in my scope.

nonsintetic, 2022-09-16:
Hello,

Thanks for the reply! The new callback will definitely be a great improvement to give some feedback about how the update is going. Going to upgrade and see how it works. 

Still interested in getting any info we can get on the progress. The device I'm working on has to run off very crappy internet in rural areas, so downloading the full update can take quite a while, the user has to keep faith that it's still doing something :)

tobozo, 2022-09-16:
There's an untested implementation (I'm still writing the CI scripts) on [this fork](https://github.com/tobozo/esp32FOTA) if you're willing to test.

Here's the code block that was added in the headers:

```C++
  // this is passed to Update.onProgress()
  typedef std::function<void(size_t, size_t)> ProgressCallback_cb; // size_t progress, size_t size
  void setProgressCb(ProgressCallback_cb fn) { _ota_progress_callback = fn; }

  // when Update.begin() returned false
  typedef std::function<void(int)> UpdateBeginFail_cb; // int partition
  void setUpdateBeginFailCb(UpdateBeginFail_cb fn) { onUpdateBeginFail = fn; }
  UpdateBeginFail_cb onUpdateBeginFail;

  // after Update.end() and before validate_sig()
  typedef std::function<void(int)> UpdateEnd_cb; // int partition
  void setUpdateEndCb(UpdateEnd_cb fn) { onUpdateEnd = fn; }
  UpdateEnd_cb onUpdateEnd;

  // validate_sig() error handling, mixed situations
  typedef std::function<void(int,int)> UpdateCheckFail_cb; // int partition, int error_code
  void setUpdateCheckFailCb(UpdateCheckFail_cb fn) { onUpdateCheckFail = fn; }
  UpdateCheckFail_cb onUpdateCheckFail;

  // update successful
  typedef std::function<void(int,bool)> UpdateFinished_cb; // int partition, bool restart_after
  void setUpdateFinishedCb(UpdateFinished_cb fn) { onUpdateFinished = fn; }
  UpdateFinished_cb onUpdateFinished;

```

PR #95 (2022-09-19): v0.2.1
- New callback setters (need tuning/feedback) fixes #87 fixes #89 :
  * `setUpdateBeginFailCb`
  * `setUpdateEndCb`
  * `setUpdateCheckFailCb`
  * `setUpdateFinishedCb`
- Arduino Library Manager compliance (now uses `#include <esp32FOTA.hpp>`) fixes #85 fixes #93 
- New constructor `esp32fota( FOTAConfig_t )` for init with predefined config, fixes #83 
- New empty constructor to use with `getConfig()` / `setConfig(FOTAConfig_t)` for late init
- Improved signature check, fixes #88 fixes #71 
- Wrapped C `semver_t` in C++ `SemverClass` 
- Added signed example to test suite
- `execOTA()` n [... cut]
closed 2022-09-19 by tobozo