chrisjoyce911/esp32FOTA #93: Arduino Library Manager missing versions

tobozo, 2022-09-16:
Just found out the Arduino Library Manager has been ignoring all versions of this library since 0.1.4 **because it contains symbolic links in the /test folder**.

I'll fix this by having the CI to copy the files instead of using the symlinks, but it's unlikely 0.1.6 and 0.2.0 and will ever appear in the Arduino Library Manager.

[image]

[image]

Platformio registry isn't affected though.

per1234, 2022-09-16:
Hi @tobozo. I think it might be challenging to set up a notification system for a couple of reasons:

- Some tags might have been created for reasons other than Library Manager publishing (e.g., beta testing, tagging revisions in unrelated branches)
- Anyone is permitted to submit libraries, so who do we notify? The library repository owner, who might have no interest in Library Manager? Or the submitter, who might not have any way to resolve the problem?

Arduino has provided an alternative solution, which is actually far superior in most ways. This is the `arduino/arduino-lint-action` GitHub Actions action. Just add a simple workflow to your repository and the action will run on every push or pull request that changes relevant files in the repository.

Arduino is using this action in all the official library, sketch, and boards platform repositories. Here is an example:

https://github.com/arduino-libraries/Arduino_PF1550/blob/master/.github/workflows/check-arduino.yml

The reason this approach is superior is because the time to catch the problem is before it is introduced into the production branch of the repository. A universal notification system as you propose could only notify of problems after they have already been released.

tobozo, 2022-09-16:
@per1234 as usual thanks for your lightning speed feedback :superhero: 

I'll use the arduino lint action as per your suggestion, the lack of notification makes perfect sense now :+1:

PR #95 (2022-09-19): v0.2.1
- New callback setters (need tuning/feedback) fixes #87 fixes #89 :
  * `setUpdateBeginFailCb`
  * `setUpdateEndCb`
  * `setUpdateCheckFailCb`
  * `setUpdateFinishedCb`
- Arduino Library Manager compliance (now uses `#include <esp32FOTA.hpp>`) fixes #85 fixes #93 
- New constructor `esp32fota( FOTAConfig_t )` for init with predefined config, fixes #83 
- New empty constructor to use with `getConfig()` / `setConfig(FOTAConfig_t)` for late init
- Improved signature check, fixes #88 fixes #71 
- Wrapped C `semver_t` in C++ `SemverClass` 
- Added signed example to test suite
- `execOTA()` n [... cut]
closed 2022-09-19 by tobozo