chrisjoyce911/esp32FOTA #94: Modify library for handling fota updates with ethernet connection instead of wifi

simogaspa84, 2022-09-16:
Hi @chrisjoyce911 ..

I am wondering if you are going to modify the library you created for handling the update with esp32 device connected with internet
with an ethernet cable.
Are you planning to do it ?

Please let me know.

Thanks

tobozo, 2022-09-16:
Hi!

There's nothing to modify if your ethernet setup uses HTTP, just replace the wifi logic by your ethernet logic in one of the examples.

simogaspa84, 2022-09-16:
Hi @tobozo .. it is http that is using wifi in the example but it is relying on the wificlientsecure class.. 
The point is to call the relative ethernet class instead the wifi client secure class..

tobozo, 2022-09-16:
You're absolutely right, however some ethernet events have `WiFiEvent_t` type so I wouldn't be surprised if all this is shared and it's fine to use WiFiClientSecure ton handle the TLS, so it might be worth a try.

[image]

[Source](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino)

If this doesn't solve the issue, you'll have to provide more info on your build:

- espressif package version
- board model
- ethernet library full name and version 
- code snippet

simogaspa84, 2022-09-18:
Hi @tobozo ..

Here some information.. 

[env:esp32dev]
platform = espressif32@3.5.0
board = esp32dev
framework = arduino

I am using currently this lib 

https://github.com/maxgerhardt/EthernetENC.git

The point is to understand where to replace in the fota lib..

tobozo, 2022-09-18:
hi @simogaspa84 

It'll probably be easier to experiment with espressif [ethernet driver](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet), maybe play with `eth_config_t` values using some of the [known registers](https://github.com/maxgerhardt/EthernetENC/blob/master/src/utility/enc28j60.h) ?

This EthernetENC library does too many things to be easily integrated, it may create huge binaries and functional redundancy with Arduino IDE because it's not specialized for esp32.

EthernetENC would probably be a better fit with esp-idf, where you can disable lwip.

simogaspa84, 2022-09-18:
Hi @tobozo .. thanks for your suggestion.. 
The point is that my ethernet shield is **enc28j60** and we already did a porting effort to make it work with esp32 and current pinout.. 
The point is that we would like to use this lib and some how to link it to the fota.
Do you think it is possible ?

https://github.com/simogaspa84/ethermet_en28j60_esp32_fw_update

Thanks

tobozo, 2022-09-18:
I'm confused with ESPAsyncServer and AsyncTCP in the lib deps, are they actually used in that sketch? 
Removing those doesn't seem to affect the build.

Also this sketch clearly shows it *is* using the WiFi lib, so you're okay with lwip and should be able tu use the esp32fota without modification.

```log
Dependency Graph
|-- EthernetENC @ 2.0.2+sha.ea07998
|   |-- SPI @ 1.0
|-- ArduinoOTA @ 1.0.9
|   |-- Update @ 1.0
|   |-- WiFi @ 1.0
|   |-- SD(esp32) @ 1.0.5
|   |   |-- FS @ 1.0
|   |   |-- SPI @ 1.0
|-- SPI @ 1.0
```

simogaspa84, 2022-09-18:
Hi @tobozo 

The example that I posted above it was just an example of an attempt.. 
The point is that the fota library is relying on the wifi as last layer.. 
Instead of using the wifi in some cases (i have to use http and the ehernet cable as physical and last layer).
The ethernet shield we used is enc28j60 and we modified the EthernetENC library to make it work with my current pinout.
I am guessing if it is possible to use this repo
 https://github.com/chrisjoyce911/esp32FOTA
but replacing the part where it uses wifi with the parts calling the Erhernetenc library.

I hope now it is more clear.
Thanks

tobozo, 2022-09-18:
> The point is that the fota library is relying on the wifi as last layer..

Actually `<WiFi.h>` presence does not necessarily imply that WiFi network will be used, but since t contains the `lwip` TCP/IP stack and makes it available to any network driver, it's unavoidable.

ex: `#include <HTTPClient.h>` will implicitely load `<WiFi.h>`

Since `lwip` is now the standard with arduino-esp32, and esp32fota is focused on the espressif version of the Arduino API, it will be a much smaller effort to adapt an Ethernet driver to the official espressif implementation than completely rewrting esp32fota just to accomodate a 2 years old orphan Ethernet API.

Here's an [Arduino compatible sketch](https://github.com/espressif/esp-idf/files/8687023/ESP32_ENC28J60.zip) for ENC28J60 Ethernet card using LWIP, it re-uses the [ENC28J60 component from esp-idf](https://github.com/espressif/esp-idf/blob/master/examples/ethernet/enc28j60) and should be a good start for integration with the espressif built-in [Ethernet](https://github.com/espressif/arduino-esp32/tree/master/libraries/Ethernet) library.

([source](https://github.com/espressif/esp-idf/issues/8376#issuecomment-1126006429))

simogaspa84, 2022-09-18:
Really @tobozo  I cannot understand why I cannot use

https://github.com/JAndrassy/EthernetENC

Can you explain to me why ?

Thanks

JeremyProffitt, 2022-09-18:
You're asking to re-write the underlying architecture, instead of hooking
into the now standard LWIP layer.

In short, you're asking for a re-write to accommodate tech debt.  And
you're welcome to do this, but the author is not willing to move in this
direction, as it is by definition, taking the whole project backwards,
leading to possible more issues with future iterations of the ESP line /
tool kit.  I would agree with this move, forward momentum is the proper
direction here.

tobozo, 2022-09-19:
@JeremyProffitt thanks for clarifying :+1: 

Some parts of the underlying architecture will have to be rewritten for what's coming next: there may be a way to provide partial support for *some* external i/o libraries (e.g. AsyncUDP, CANBus, Serial, etc) while still moving the project forward.

@simogaspa84 I'll soon experiment gz/targz support with esp32fota using Arduino `Stream`, which should offer a better perspective for integrating other network libraries (gz/targz updating needs to override the default http/client logic too).

This will happen on another fork, I'll ping this thread so you can be notified about the progress.

simogaspa84, 2022-09-25:
So @JeremyProffitt and @tobozo  summarizing which is the best solution for you for proceeding?
Which is the starting point for doing the porting?
My task is always the same making a fota update which works for the ethernet shield enc28j60.
I would to hear from you which are the basic steps and the starting library to use.
Thanks a lot for your time

JeremyProffitt, 2022-09-25:
@simogaspa84 Sounds like you need to start testing the enc28j60 driver with lwip. Perhaps this library https://github.com/matwey/lwip-enc28j60 - though it looks pretty generic, which to me means low performance.  And start looking at strategies to port this library to support lwip as well if it doesn't.

Unfortunately, there is no free lunch here, I'd start by testing an lwip implementation for your ethernet and making that rock solid and performant, before anything else.

simogaspa84, 2022-09-26:
If you can help me when you have free time @JeremyProffitt  I really appreciate it.. I think the fota lib supporting also some ethernet shields would be a super boost.
Thanks a lot

tobozo, 2022-09-27:
@simogaspa84 

I've experimented with my ENC28J60 module and had good results with HTTPS, but I had to bake my own library, maybe you'll find it useful:  https://github.com/tobozo/ESP32-ENC28J60

simogaspa84, 2022-09-28:
Thanks @tobozo .. I will try the library too. and it it works also for me then we can use it or call it in esp32fota?
Correct?

tobozo, 2022-09-28:
@simogaspa84 not entirely, there's still a [bit of logic](https://github.com/chrisjoyce911/esp32FOTA/blob/master/src/esp32FOTA.cpp#L688) that needs to be replaced by the ethernet counterpart.

You'll have to comment out those three lines for the meantime and make sure the `if( eth_connected )` test is performed in your `setup()`.

I'm still testing the stream variants (http/file/serial) but I had to break some eggs and it made the binary size 3% bigger, it's not stable enough to be pushed yet.

tobozo, 2022-10-01:
@simogaspa84 did you get some results with [ESP32-ENC28J60](https://github.com/tobozo/ESP32-ENC28J60)?

The latest release of esp32FOTA allows some logic overriding for the connection checker.

You can now use a function declared in your sketch and attach it to esp32FOTA to replace the WiFi check during the operations:

```cpp
static bool EthernetStatusCheck()
{
    return eth_connected;
}

void setup()
{
    // ...
    FOTA.setStatusChecker( EthernetStatusCheck );
}

void loop()
{
    FOTA.handle();
}

```

tobozo, 2022-10-01:
now there's an [ENC28J60 + esp32FOTA example](https://github.com/tobozo/ESP32-ENC28J60/tree/main/examples/FOTA)

simogaspa84, 2022-10-03:
you are my hero @tobozo .. thanks

simogaspa84, 2022-10-11:
Hi @tobozo  i am trying your new example here [ENC28J60 + esp32FOTA example](https://github.com/tobozo/ESP32-ENC28J60/tree/main/examples/FOTA)

But when i import the files and folders under platformio with arduino framework it shows that many files are missing ..

esp_eth_phy.h
esp_eth_mac.h
eth_phy_regs_struct.h
esp_eth_enc28j60.h
esp_netif.h
esp_eth_com.h
soc/emac_ext_struct.h

Can you tell me how you tried the lib ? 

Thanks a lot

tobozo, 2022-10-11:
I replied on the other issue you created.

closing this as solved

closed 2022-10-11 by tobozo