chrisjoyce911/esp32FOTA #99: [E][HTTPClient.cpp:251] beginInternal(): failed to parse protocol

simogaspa84, 2022-09-26:
Hi.. I created a web hosting account here.. 

https://systelfota.000webhostapp.com/

I am trying to update the fw after I have created the json file and uploaded with the .bin.. 

Running the example with https but not certificate I get 
`
 [E][HTTPClient.cpp:251] beginInternal(): failed to parse protocol`

Can you tell me the meaning of this error ? 

Thanks 

`#include <esp32fota.h>
#include <WiFi.h>

// Change to your WiFi credentials
const char *ssid = "";
const char *password = "";

// esp32fota esp32fota("<Type of Firme for this device>", <this version>, <validate signature>);
esp32FOTA esp32FOTA("esp32-fota-http", 1, false, true);

void setup_wifi()
{
  delay(10);
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println(WiFi.localIP());
}

/**
 * @brief  Function to execute the setup generated by the arduino framework
 * @param  None
 * @retval None
 */
void setup(void)
{
  esp32FOTA.setManifestURL("https://systelfota.000webhostapp.com/fota.json");
  Serial.begin(115200);
  setup_wifi();
}
/**
 * @brief  Function to execute in loop generated by the arduino framework
 * @param  None
 * @retval None
 */
void loop(void)
{
  bool updatedNeeded = esp32FOTA.execHTTPcheck();
  if (updatedNeeded)
  {
    esp32FOTA.execOTA();
  }

  delay(2000);
}`

tobozo, 2022-09-26:
thanks for your report, looks like a bug in the library code

until this is fixed, one way to avoid the bug is to use a define or static variable to store the URL.

```cpp

String my_manifest_url = "https://systelfota.000webhostapp.com/fota.json";

void setup(void)
{
  esp32FOTA.setManifestURL( my_manifest_url );
  Serial.begin(115200);
  setup_wifi();
}

```

simogaspa84, 2022-09-27:
Thanks @tobozo for your suggestion.. I will try it now.. I am not sure that everything is set correctly  So i put here the json .. 

`{
    "type": "esp32-fota-http",
    "version": "2",
    "url": "https://systelfota.000webhostapp.com/update.bin"
}`

Can you check?

Thanks a lot

simogaspa84, 2022-09-27:
@tobozo  please can you suggest a web hosting provider easy to configure to try all examples in your lib?
Thanks a lot

tobozo, 2022-09-27:
@simogaspa84 your config looks good, can you try this sketch?

```cpp

#include <esp32fota.h> // fota pulls WiFi library

const char *ssid = "zzzzz";
const char *password = "xxxx";

const char* firmware_name   = "esp32-fota-http";
const bool check_signature  = false;
const bool disable_security = true;

#define FOTA_URL "https://systelfota.000webhostapp.com/fota.json"

int firmware_version_major  = 2;
int firmware_version_minor  = 0;
int firmware_version_patch  = 0;

esp32FOTA FOTA; // empty constructor

void setup_wifi()
{
  delay(10);
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println(WiFi.localIP());
}

void setup(void)
{
  Serial.begin(115200);
  {
    auto cfg = FOTA.getConfig();
    cfg.name         = firmware_name;
    cfg.manifest_url = FOTA_URL;
    cfg.sem          = SemverClass( firmware_version_major, firmware_version_minor, firmware_version_patch );
    cfg.check_sig    = check_signature;
    cfg.unsafe       = disable_security;
    //cfg.root_ca      = MyRootCA;
    //cfg.pub_key      = MyRSAKey;
    FOTA.setConfig( cfg );
  }
  setup_wifi();
}

void loop(void)
{
  bool updatedNeeded = FOTA.execHTTPcheck();
  if (updatedNeeded)
  {
    FOTA.execOTA();
  }
  delay(2000);
}

```

simogaspa84, 2022-09-27:
hi @tobozo .. thanks for your code.. 

I have tried but it is failing   

`entry 0x400806a8
Connecting to Systel-Guest
.....
172.16.10.67
Error on HTTP request (httpCode=404)
Error on HTTP request (httpCode=404)
Error on HTTP request (httpCode=404)
Error on HTTP request (httpCode=404)
Error on HTTP request (httpCode=404)
Error on HTTP request (httpCode=404)
Error on HTTP request (httpCode=404)
Error on HTTP request (httpCode=404)`

tobozo, 2022-09-27:
maybe give it a domain name instead of an IP address when calling HTTPS urls, or just use plain HTTP ? :wink:

simogaspa84, 2022-09-28:
Sorry @tobozo .. I don't understand.. which is the mistake? 
What should i do or change?
Thanks again

simogaspa84, 2022-09-28:
Hi @tobozo do i have to do something in the web hosting portal?
Can i keep the code in the same way?

Thanks a lot for your kind reply

tobozo, 2022-09-28:
I don't know but 172.16.10.67 seems like a C class address (private) and those can't be routed on the internet, the domain name you specified earlier resolves to 145.14.145.212 and I don't see it in the logs, so maybe the failure is a consequence of that.

simogaspa84, 2022-09-28:
ok @tobozo so what do you suggest to do ? 
Thanks

vortigont, 2022-09-29:
BTW, I've faced same issues. Looks like constructor with String object as input parameter turns into dangling pointer with const char * in cfg struct.
A suggested fix is to use const char overloads and pass String obj by reference - fix is included in PR #100

tobozo, 2022-09-29:
indeed that was goofy, sorry about that :bow: 

looking at your PR: I didn't know this notation, how does it compare to `String blah="";` ?

```cpp
    String UpdateURL((char *)0);
    String PartitionLabel((char *)0);
```

vortigont, 2022-09-29:
`String UpdateURL((char *)0);`
this could be useful sometimes if you need a real empty String obj.

`String blah="";` creates the obj with default ctor, than assigns a rvalue "" to it (which is const char*).
A default String preallocates some memory in advance (if that has not been changed recently).
And default c-tor for String is String(const char *cstr = ""); :))) So it's just unnecessary duplication, `String blah;` will end up with the same.

`String UpdateURL((char *)0);` creates String obj using c-tor with zero byte argument, so it ends up creating a really empty String in-place using move under the hood (as I remember it).

Mostly useful if combined with `strobj.reserve()` later to avoid extra mem allocation on object creation.

simogaspa84, 2022-09-30:
thanks @tobozo and @vortigont

tobozo, 2022-09-30:
@simogaspa84 there's a better implementation and more fixes from @vortigont on the [0.2.3](https://github.com/chrisjoyce911/esp32FOTA/tree/0.2.3) branch

commit a42c7625 (2022-09-30): store manifest url locally, fixes #99
closed 2022-09-30 by tobozo