chrisjoyce911/esp32FOTA #105: HTTp error..

simogaspa84, 2022-10-11:
Hi @tobozo .

I am trying the new branch but still have this issue .. 

Opening item http://192.168.0.100/fota/esp32-fota-http-2.bin
[E][ssl_client.cpp:98] start_ssl_client(): Connect to Server failed!
[E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -1

[image]

This is the sketch

`#include <esp32fota.h> // fota pulls WiFi library

const char *ssid = "769D4F";
const char *password = "EVW32C0N00028997";

const char *firmware_name = "esp32-fota-http";
const bool check_signature = false;
const bool disable_security = true;

#define FOTA_URL "https://systelfota.000webhostapp.com/fota.json"

int firmware_version_major = 2;
int firmware_version_minor = 0;
int firmware_version_patch = 0;

esp32FOTA FOTA; // empty constructor

void setup_wifi()
{
  delay(10);
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println(WiFi.localIP());
}

void setup(void)
{
  Serial.begin(115200);
  {
    auto cfg = FOTA.getConfig();
    cfg.name = firmware_name;
    cfg.manifest_url = FOTA_URL;
    cfg.sem = SemverClass(firmware_version_major, firmware_version_minor, firmware_version_patch);
    cfg.check_sig = check_signature;
    cfg.unsafe = disable_security;
    // cfg.root_ca      = MyRootCA;
    // cfg.pub_key      = MyRSAKey;
    FOTA.setConfig(cfg);
  }
  setup_wifi();
}

void loop(void)
{
  bool updatedNeeded = FOTA.execHTTPcheck();
  if (updatedNeeded)
  {
    FOTA.execOTA();
  }
  delay(2000);
}`

tobozo, 2022-10-11:
hi,

Thanks for your report, please add more precision e.g. what happened before the error, also provide source of the "other" firmware.

#### setting arduino debug level to `debug` and providing a full log would be very helpful as it'd show the full scenario.

Meanwhile all I can do is test the URL's, so far the web server response looks good for the manifest:

```log
curl -i https://systelfota.000webhostapp.com/fota.json
HTTP/2 200 
date: Tue, 11 Oct 2022 09:06:17 GMT
content-type: application/json
content-length: 119
last-modified: Tue, 11 Oct 2022 09:03:48 GMT
accept-ranges: bytes
server: awex
x-xss-protection: 1; mode=block
x-content-type-options: nosniff
x-request-id: d27099d654a1174f7cd21c4b6329b75b

{
    "type": "esp32-fota-http",
    "version": "2",
    "url": "https://systelfota.000webhostapp.com/update.bin"
}
```

The response also looks good for the firmware:

```log
curl -I https://systelfota.000webhostapp.com/update.bin
HTTP/2 200 
date: Tue, 11 Oct 2022 09:05:52 GMT
content-type: application/octet-stream
content-length: 318464
last-modified: Tue, 27 Sep 2022 15:33:34 GMT
accept-ranges: bytes
server: awex
x-xss-protection: 1; mode=block
x-content-type-options: nosniff
x-request-id: 92f37fae6b3b804c43b35b77cf1de1c0
```

Speculation: could it be that it actually worked, the **target firmware** loaded successfully but is an old firmware still pointing to your initial test environment? 
Using a `Serial.begin(115200);Serial.println("hello world");` sketch as the target firmware (no esp32-fota in there) could dissipate any doubt about that.

Observation: your screenshot shows a `public_html` folder, does it come with the hosting? If so it's likely everything outside this folder is ignored and not made visible by the web server, you should probably delete those three weeks old manifest and firmware files to make sure you're not hitting the past :wink:

simogaspa84, 2022-10-11:
Hi @tobozo .. thanks for your super help... now it works... 

this is the final example working...   

`
#include <esp32fota.h> // fota pulls WiFi library

// Change to your WiFi credentials
const char *ssid = "769D4F";
const char *password = "EVW32C0N00028997";

const char *firmware_name = "esp32-fota-http";
const bool check_signature = false;
const bool disable_security = true;

#define FOTA_URL "https://systelfota.000webhostapp.com/fota.json"

int firmware_version_major = 2;
int firmware_version_minor = 0;
int firmware_version_patch = 0;

esp32FOTA FOTA; // empty constructor

void setup_wifi()
{
  delay(10);
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println(WiFi.localIP());
}

void setup(void)
{
  Serial.begin(115200);
  {
    auto cfg = FOTA.getConfig();
    cfg.name = firmware_name;
    cfg.manifest_url = FOTA_URL;
    cfg.sem = SemverClass(firmware_version_major, firmware_version_minor, firmware_version_patch);
    cfg.check_sig = check_signature;
    cfg.unsafe = disable_security;
    // cfg.root_ca      = MyRootCA;
    // cfg.pub_key      = MyRSAKey;
    FOTA.setConfig(cfg);
  }
  setup_wifi();
}

void loop(void)
{
  bool updatedNeeded = FOTA.execHTTPcheck();
  if (updatedNeeded)
  {
    FOTA.execOTA();
  }
  delay(2000);
}

Just 2 questions... 

1- Now it is working with https without checking the certificate... which is the next step to add the check of the certificate.. 
2-Can you post again the lin kto the repo with the the fota example for ethernet ?? Thanks a lot @tobozo 
`

PR #117 (2022-12-08): 0.2.6
- Removed `esp32FOTA::checkURL` and `esp32FOTA::useDeviceID`
- Improved memory management with config
- Added CI Tests for Arduino IDE
- Fixed `withDeviceID.ino` example

fixes #105 
fixes #115 
fixes #116
closed 2022-12-08 by tobozo