chrisjoyce911/esp32FOTA #106: esp32FOTA doesn't work with Bluetooth SerialBT

DonatelloX, 2022-10-16:
**Arduino board:**  _ESP32 Dev kit_
**Arduino IDE:** _2.00_
**Espressif Arduino core for the ESP32:** _2.0.6_

**Arduino Sketch:**
`#include "BluetoothSerial.h"
#include <stdio.h>
#include <stdlib.h>
#include <Wire.h>
#include <WiFi.h>
#include <SPIFFS.h>
#include <esp32fota.h>    

BluetoothSerial SerialBT;

#define FW_URL_UPGRADE        "https://www.mysite.com/fota/fota.json" 

// Change to your WiFi credentials
const char *ssid = "MySSID";
const char *password = "P@ssword";

// esp32fota esp32fota("<Type of Firmware for this device>", <this version>, <validate signature>, <allow insecure https>);
esp32FOTA esp32FOTA("esp32-fota-http", 100, false, true);

void setup_wifi()
{
  delay(3000);
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println(WiFi.localIP());
}

void setup()
{
  Wire.begin();
  Wire.setClock();

  Serial.begin(115200);
  setup_wifi();
}

void loop()
{
  SerialBT.begin("BLE NAME");
  esp32FOTA.checkURL = FW_URL_UPGRADE;
  bool updatedNeeded = esp32FOTA.execHTTPcheck();
  if (updatedNeeded)
  {
    esp32FOTA.execOTA();
  } else {
    Serial.println("no update");
  }

  while(1);
}`

**Problem:**
If I use _SerialBT.begin("BLE NAME")_ (please see the code above) BEFORE the _esp32FOTA.execHTTPcheck()_ has problems with heap stack space. If I use _SerialBT.begin("BLE NAME")_ (please see the code above) AFTER the _esp32FOTA.execHTTPcheck()_, Bluetooth and WiFi are not working.

Please, how can you fix this?

tobozo, 2022-10-16:
hi thanks for your feedback,

> Espressif Arduino core for the ESP32: 2.0.6

can't find this version, the highest currently available is 2.0.5

some thoughts:

don't include BluetoothSerial with quotes but with brackets:
```cpp
#include <BluetoothSerial.h>
```

you don't need that, esp32fota pulls the wifi library
```cpp
#include <stdio.h>
#include <stdlib.h>
#include <Wire.h>
#include <WiFi.h>
```

this should be in your setup() not in your loop()
```cpp
SerialBT.begin("BLE NAME");
```

Have you tried LittleFS instead of SPIFFS? uses the same syntax and may solve some memory issues
```cpp
#include <SPIFFS.h>
```

closed 2022-10-17 by tobozo

DonatelloX, 2022-10-17:
I have found a possible solution:
_Use esp32FOTA **1.0.6** version downloaded from the repository and manually installed in Arduino._
I have to test the solution well but now there seems to be no memory problem.

DonatelloX, 2022-10-24:
@chrisjoyce911 Please, can you add version 0.1.6 of esp32FOTA in Arduino Library? This version solves my memory problems and works with https without root.
Thank you

tobozo, 2022-10-24:
[see this issue](https://github.com/chrisjoyce911/esp32FOTA/issues/93) the 0.1.6 contains a symlink

tobozo, 2022-10-24:
I'll try to delete and re-create the tag, wish me luck :)

tobozo, 2022-10-24:
[image]