chrisjoyce911/esp32FOTA #107: Cowardly refusing to overwrite U_SPIFFS

HornyPriest, 2022-10-28:
Hello,

I have a problem with using esp32FOTA to update spiffs.bin.gz file from server.
I am very sorry if this is not a place for my issue. In that case please delete this issue.

- Arduino board:  esp32-wroom-32u

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.19

Problem is i can not update spiffs.bin file from server.
Esp32 returns error code with message: _Cowardly refusing to overwrite U_SPIFFS with http://lockit.pro:80/ota/Dinamics/spiffs.bin.gz. Use setCertFileSystem(nullptr) along with setPubKey()/setCAPem() to enable this feature._

I have been able to update firmware.bin with whis library though.
Have anyone had similar issue or any idea how should i resolve this.

Thank you for helping me.

My code without unnecessary stuff in looks like this:

```
#include "EmonLib.h" // Include Emon Library
#include "math.h"
#include "driver/adc.h"
#include "esp32FOTA.hpp"
#include "WiFi.h"
#include "PubSubClient.h"
#include "ESPAsyncWebServer.h"            // https://github.com/me-no-dev/ESPAsyncWebServer
// download zip from above->Arduino IDE->menu->tab->sketch->include library->add ZIP library
#include "AsyncTCP.h"                     // https://github.com/me-no-dev/AsyncTCP
// download zip from above->Arduino IDE->menu->tab->sketch->include library->add ZIP library
#include "SPIFFS.h"
//#include "LittleFS.h"
#include "ESPmDNS.h"
#include "HTTPClient.h"
#include "ESP32httpUpdate.h"
#include "Preferences.h"
#include "ESP32-targz.h" // http://github.com/tobozo/ESP32-targz

// Initiate Preferences to save WiFi credentials to EEPROM
Preferences preferences;
boolean SavedWiFi = LOW;
boolean SavedCalibration = LOW;

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);

// Change to your WiFi credentials
//const char *ssid = "Implera01";
//const char *password = "********";

// esp32fota esp32fota("<Type of Firme for this device>", <this version>);
esp32FOTA esp32FOTA;

int FW_version = 3;
String FW_versionStr = "0.3a";

#define FOTA_URL "http://lockit.pro/ota/Dinamics/Dinamics.json"
const char *firmware_name = "Dinamics";
const bool check_signature = false;
const bool disable_security = true;

int firmware_version_major = 3;
int firmware_version_minor = 0;
int firmware_version_patch = 0;

WiFiClient espClient;
PubSubClient client(espClient);

void initSPIFFS() {
  if (!SPIFFS.begin(true)) {
    Serial.println("An error has occurred while mounting SPIFFS");
  }
  Serial.println("SPIFFS mounted successfully");
}

void setup() {
  // put your setup code here, to run once:

  Serial.begin(115200);
  delay(200);
  
  
  initSPIFFS();

  {
    auto cfg = esp32FOTA.getConfig();
    cfg.name = firmware_name;
    cfg.manifest_url = FOTA_URL;
    cfg.sem = SemverClass(firmware_version_major, firmware_version_minor, firmware_version_patch);
    cfg.check_sig = check_signature;
    cfg.unsafe = disable_security;
    // cfg.root_ca = MyRootCA;
    // cfg.pub_key = MyRSAKey;
    esp32FOTA.setConfig(cfg);
  }

    if (initWiFi()) {
   //  webpage code
   }
 
  }

  
  vTaskDelay(300);
  client.setServer(mqtt_server, 31883);
  client.setCallback(callback);

  //init and get the time
  configTime(0, 0, ntpServer);

  digitalWrite(LED_GREEN, LOW);
  digitalWrite(LED_RED, LOW);
  
  SetupComplete = HIGH;
}

void loop(){
if(UpdateStart == HIGH){
    if((WiFi.status() == WL_CONNECTED)) {
      boolean updatedNeeded = esp32FOTA.execHTTPcheck();
      if(updatedNeeded == HIGH){
        Serial.print("najden update");
        server.end();
        esp32FOTA.execOTA();
      }else{
        UpdateStart = LOW;
      }
    }
  }

  if((WiFi.status() == WL_CONNECTED)){
    client.loop();
    if(UpdateSpiffs == HIGH) {
        Serial.println("checking for wp update");
        boolean updatedNeeded = esp32FOTA.execHTTPcheck();
        if(updatedNeeded == HIGH){
          server.end();
//          SPIFFS.end();
          SPIFFS.format();
          Serial.println("Update SPIFFS...");
          debug += "$";
          debug += "SPIFFS update select";
          debug += "$";
          esp32FOTA.execOTA();
          UpdateSpiffs = LOW;
        }else{
          UpdateSpiffs = LOW;
        }
    }else{
      if(!SPIFFS.exists("/Implera-Dynamics.html.gz")){
        UpdateSpiffs = HIGH;
      }
    }
  }
}

```

My JSON file looks like this:

`{
    "type": "Dinamics",
    "version": "4.0.0",
    "host": "lockit.pro",
    "port": 80,
    "bin": "/ota/Dinamics/DinamicsV4.bin",
    "spiffs": "/ota/Dinamics/spiffs.bin.gz"
}`

tobozo, 2022-10-28:
hi thanks for your feedback

did you try to run this after setConfig() as suggested by the error message?
 
```cpp
esp32FOTA.setCertFileSystem(nullptr)
```

nichwall, 2022-11-02:
I had the same issue. Using only
```
esp32FOTA.setCertFileSystem(nullptr)
```
allowed SPIFFS to update correctly, but the error message implies the `setPubKey()` or `setCAPem()` functions need to be called, which seems to be incorrect because I then cannot compile the project. Are one of these functions also needed, or can they be ignored? I can open a new issue if you want to go more in depth.

HornyPriest, 2022-11-02:
> hi thanks for your feedback
> 
> did you try to run this after setConfig() as suggested by the error message?
> 
> ```c++
> esp32FOTA.setCertFileSystem(nullptr)
> ```

Thank you very much. I needed to add that line in mu vode, but be careful to pot in in {} brackets right after .setConfig

tobozo, 2022-11-07:
@nichwall `esp32Fota.setPubKey( you_pub_key )` and  `esp32Fota.setCAPem( your_ca_pem )` are only necessary if your setup uses certificates and/or signature check, they just can't be stored on SPIFFS/LittleFS if that partition is also a target for update.

@HornyPriest I'm happy it worked, closing this as fixed

closed 2022-11-07 by tobozo