chrisjoyce911/esp32FOTA #108: FOTA url converter to const char *

simogaspa84, 2022-11-17:
Hi @chrisjoyce911 ..

I have alredy used your incredible lib.
My request is a doubt not as issue just from the side of the lib..
Instead of putting the fota url in the code i am parsing it from a message to the cloud.
The point is that also if the content it is the same I got failed to parse protocol..

String fota_url is the url to connect and it is a  type String

When I try to print the following

  Serial.println(fota_url);
  Serial.println(fota_url.c_str());

#define FOTA_URL fota_url.c_str()
    Serial.println(FOTA_URL);

Are always the same for example

http://amazon...xxxx

but instead if i do directly with

#define FOTA_URL "http://amazon...xxxx"

it is working ... 

Do you know why?

Thanks a lot

tobozo, 2022-11-17:
maybe use this:

```cpp
static String fota_url;
``` 
instead of `String fota_url;`

simogaspa84, 2022-11-18:
thanks @tobozo

closed 2022-11-18 by simogaspa84