chrisjoyce911/esp32FOTA #109: Using esp32FOTA with wt32-eth01

zjonesz, 2022-11-21:
Is it possible to run this library with wired ethernet?
Would be nice to use a wt32-eth01 😁

tobozo, 2022-11-21:
yup it's possible and it's very fast (5 seconds OTA)

this highly depends on the latest version of arduino-esp32 core though, here's the code:

```cpp
#define ETH_PHY_TYPE ETH_PHY_LAN8720
#include <ETH.h>
static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case ARDUINO_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case ARDUINO_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case ARDUINO_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case ARDUINO_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case ARDUINO_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

#include <HTTPClient.h>
#include <WiFiClientSecure.h>

HTTPClient http;
WiFiClientSecure client;

void testClient(const char * url)
{
  Serial.printf("\nConnecting to %s\n", url);

  http.setFollowRedirects(HTTPC_STRICT_FOLLOW_REDIRECTS);

  if( String(url).startsWith("https") ) {
    client.setInsecure();
    http.begin( client, url );
  } else {
    http.begin( url );
  }

  int httpCode = http.GET();

  if( httpCode != HTTP_CODE_OK && httpCode != HTTP_CODE_MOVED_PERMANENTLY ) {
    Serial.printf("Error: Status %i\n", httpCode );
  }

  Stream* stream = http.getStreamPtr();

  while (stream->available()) {
    Serial.write(stream->read());
  }

  Serial.println("closing connection\n");
  http.end();
}

void setup()
{
  Serial.begin( 115200 );
  WiFi.onEvent( WiFiEvent );
  ETH.begin();

  while( !eth_connected) {
    Serial.println("Connecting...");
    delay( 1000 );
  }

  testClient("https://google.com");

}

void loop()
{
}

```

tobozo, 2022-12-06:
Also see [@khoih-prog](https://github.com/khoih-prog)'s [WebServer_ESP32_ENC](https://github.com/khoih-prog/WebServer_ESP32_ENC) and [AsyncWebServer_ESP32_ENC](https://github.com/khoih-prog/AsyncWebServer_ESP32_ENC).

Closing this as answered.

closed 2022-12-06 by tobozo