chrisjoyce911/esp32FOTA #110: Signature Verification at the beginning not at the end

simogaspa84, 2022-11-24:
Hi @chrisjoyce911 ,, 
thanks again for your great work,,

It is not a bug just a request.. 
I have just tried the fota with signature check and it works perfectly.

Just a question.. is it too complicated or not possible to check if the signature is correct at the beginning and not at the end?

Please let me know.. 

Thanks

tobozo, 2022-11-24:
@simogaspa84 hash is verified by reading every byte of the binary, so the whole file is needed and the result can only be known at the end.

The `fail only after downloading everything` situation is by design indeed.

If you want to verify a partition binary before `Update.begin()` (but why would you want to do it?), you need a filesystem big enough to fit the downloaded partition (e.g. SD Card, or huge SPIFFS), and implement your own logic for transaction, rollback and history, none of which are part of the esp32FOTA scope.

Feel free to continue the discussion here, closing the issue as out of scope but keeping the conversation as insightful.

closed 2022-11-24 by tobozo

simogaspa84, 2022-11-25:
Hi @tobozo  

@simogaspa84 hash is verified by reading every byte of the binary, so the whole file is needed and the result can only be known at the end.

Thanks for your perfect answer.