chrisjoyce911/esp32FOTA #111: setManifestURL appears to not work without checkURL being set?

Humancell, 2022-11-28:
Hello,

What a great project!  And I did get it working, but not how I expected.  I've used Arduino for a long time and written lots of code and am not sure if there is something I am doing wrong, or if there is a bug.

- Arduino board:  ESP32-WROOM-32 Dev Board

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.19

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): 

I've followed your examples and included the header and created the object:

```
/*
 * For the OTA update support
 * Note: This was taken from a esp32FOTA example
 */
#include <esp32fota.h>
#include <WiFi.h>

/*
 * We're going to be using WiFi
 * TODO: Add the WiFi Manager library to allow dynamic configuration
 * 
 */
const char *ssid = "myWiFi";
const char *password = "myPass";

/*
 * Create the esp32FOTA object
 * NOTE:  This sets the firmware type and version for this code
 */
esp32FOTA esp32FOTA("esp32-with-OTA", "1.4.0", false);
```
The file on my server is:

```
[
{
  "type":"esp32-ble-advertisement-scanner-with-OTA",
  "version":"1.4.1",
  "url":"http://myServer.myDomain.com/esp32fota/esp32-ble-advertisement-scanner-with-OTA.1.4.1.bin"
},
{
  "type":"esp32-with-OTA",
  "version":"1.4.1",
  "url":"http://myServer.myDomain.com/esp32fota/esp32-with-OTA.1.4.1.bin"
}
]
```
In set-up if I have this:

```
  esp32FOTA.setManifestURL("http://myserver.mydomain.com/esp32fota/esp32FOTA-test.json");
  esp32FOTA.useDeviceId( true ); // optionally append the device ID to the HTTP query
//  esp32FOTA.checkURL = "http://myserver.mydomain.com.com/esp32fota/esp32FOTA-test.json";
```
Then when I run this in Loop:

```
 if(cold_boot){
    Serial.println("============================================================================");
    Serial.println("Performing FOTA Update Check ...");
    // esp32FOTA.handle();
    // or ...
    bool updateNeeded = esp32FOTA.execHTTPcheck();
    Serial.printf("updateNeeded: %s\n", updateNeeded ? "true" : "false");
    if (updateNeeded) {
      Serial.println("Update needed!");
      //   esp32FOTA.execOTA();
    } else {
      Serial.println("No update needed!");
    }
    //delay(2000);
    cold_boot = false;
  }
```
I see this is the debug output:

```
Performing FOTA Update Check ...
[  4105][I][esp32FOTA.cpp:709] execHTTPcheck(): Getting HTTP: ?id=255127568175012
[  4105][I][esp32FOTA.cpp:710] execHTTPcheck(): ------
[  4110][I][esp32FOTA.cpp:302] setupHTTP(): Connecting to: ?id=255127568175012

[  4117][E][HTTPClient.cpp:257] beginInternal(): failed to parse protocol
[  4124][E][HTTPClient.cpp:257] beginInternal(): failed to parse protocol
[  4130][W][HTTPClient.cpp:1469] returnError(): error(-1): connection refused
updateNeeded: false
No update needed!
```
And it fails.  The ESP32 never even touches my webserver.

If I then uncomment the "checkURL" line:

```
  esp32FOTA.setManifestURL("http://myServer.myDomain.com/esp32fota/esp32FOTA-test.json");
  esp32FOTA.useDeviceId( true ); // optionally append the device ID to the HTTP query
  esp32FOTA.checkURL = "http://myServer.myDomain.com/esp32fota/esp32FOTA-test.json";
```
The I see this in the debug output:

```
Performing FOTA Update Check ...
[  3573][I][esp32FOTA.cpp:709] execHTTPcheck(): Getting HTTP: http://myServer.myDomain.com/esp32fota/esp32FOTA-test.json?id=255127568175012
[  3577][I][esp32FOTA.cpp:710] execHTTPcheck(): ------
[  3582][I][esp32FOTA.cpp:302] setupHTTP(): Connecting to: http://myServer.myDomain.com/esp32fota/esp32FOTA-test.json?id=255127568175012

[  3775][I][esp32FOTA.cpp:603] checkJSONManifest(): Payload type in manifest esp32-with-OTA matches current firmware esp32-with-OTA
[  3776][I][esp32FOTA.cpp:830] debugSemVer(): Payload firmware version: 1.4.1
[  3783][I][esp32FOTA.cpp:654] checkJSONManifest(): JSON manifest provided keys: url=true, host: false, port: false, bin: false, fs: []
updateNeeded: true
Update needed!
```

And it succeeds!  The hit on my webserver is visible in the logs, and it worked.

I found the checkURL in your example here: https://github.com/chrisjoyce911/esp32FOTA/blob/master/examples/HTTP/HTTPS_without_root_cert.ino

Why is the "checkURL" assignment required?

Humancell, 2022-11-28:
I might have found my own answer?

In looking through the code I notice that several of the methods are commented as "deprecated".  Is it the intention that all configuration should move to the "late" model in setup?

```
  {
    auto cfg = FOTA.getConfig();
    cfg.name          = "esp32-fota-http";
    cfg.manifest_url  = "http://server/fota/fota.json";
    cfg.sem           = SemverClass( 1, 0, 0 ); // major, minor, patch
    cfg.check_sig     = false; // verify signed firmware with rsa public key
    cfg.unsafe        = true; // disable certificate check when using TLS
    //cfg.root_ca       = MyRootCA;
    //cfg.pub_key       = MyRSAKey;
    //cfg.use_device_id = false;
    FOTA.setConfig( cfg );
  }
```
I'm testing with this now to see if this works cleanly.

tobozo, 2022-11-29:
@Humancell `name` and `manifest_url` config properties are pointers, so you'll have better results by storing the string values as `#define my_manifest_url "..."` or `static const char* my_manifest_url = "...";` before assigning them.

Humancell, 2022-12-01:
@tobozo thanks for that tip.  I was following the examples and documentation which didn't appear to use that.

I've now migrated to using the "late" config above, and when I implemented that I did end up using a const and #define for the firmware type and version.

It all seems to be working great now!

closed 2022-12-01 by Humancell

tobozo, 2022-12-01:
@Humancell 

>  I was following the examples and documentation 

thanks for mentioning that, I've updated both documentation and examples, will publish a new release soon