chrisjoyce911/esp32FOTA #113: Version 0.2.4 does not compile in Arduino IDE 2.01 for ESP32

buttim, 2022-12-05:
- Arduino board:  TBeam

- Arduino IDE version (found in Arduino -> About Arduino menu):  2.01

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too):
. include ESP32Fota.h
. compile
Compilation error:
esp32FOTA.cpp:506:93: error: call of overloaded 'String(int64_t&)' is ambiguous

tobozo, 2022-12-05:
Just tried with Arduino IDE 2.0.3 and got no such error, even when setting the compiler warnings to "All".

buttim, 2022-12-05:
Updated my IDE to 2.0.3 and the error is still present. If I modify line 506 introducing a cast to `int` for the variable `updateSize` the error disappears. The modified line reads as follows:
`        Serial.println("Written only : " + String(written) + "/" + String((int)updateSize) + ". Premature end of stream?");
`

buttim, 2022-12-05:
Forgot to mention I'm using a Windows machine. I don't know if that might be relevant or not

tobozo, 2022-12-06:
thanks for these precisions, I've pushed a fix and will produce a release sonn

closed 2022-12-06 by tobozo

buttim, 2022-12-06:
Thank you so much!

tobozo, 2022-12-06:
well thanks to you for pointing that out :+1:

tobozo, 2022-12-08:
fixed in [0.2.6](https://github.com/chrisjoyce911/esp32FOTA/releases/tag/v0.2.6), will propagate soon in the Arduino Library Manager