chrisjoyce911/esp32FOTA #116: Error while compiling HTTP example

milkywade, 2022-12-07:
Hi - using last version 0.2.5, trying to compile the following example:

https://github.com/chrisjoyce911/esp32FOTA/blob/master/examples/HTTP/HTTP.ino

against this fota

```
{
    "type": "test_1",
    "version": "1",
    "url": "http://192.168.1.100/ota/test_1-version-1.bin"
}
```

returns the following message on the terminal (using platform.io):

```
[353940][E][HTTPClient.cpp:257] beginInternal(): failed to parse protocol
[353940][E][HTTPClient.cpp:257] beginInternal(): failed to parse protocol
```

P.S. the command executed (once) is
`bool updatedNeeded = esp32FOTA.execHTTPcheck();`

milkywade, 2022-12-07:
taking the "s from the version number results in the same error

milkywade, 2022-12-07:
Tried several changes from the HTTP example, with no success:

1. changing the include statement to (as per the example on the main github page)
`#include <esp32fota.hpp>`

2. changing the version to a string, with three sub-versions, in this line:
`esp32FOTA esp32FOTA("cave_1", "1.0.0");`

3. using the `esp32FOTA.handle();` shortcut in the `loop()` section

...any ideas ?

tobozo, 2022-12-08:
fixed in [0.2.6](https://github.com/chrisjoyce911/esp32FOTA/releases/tag/v0.2.6), will propagate soon in the Arduino Library Manager

PR #117 (2022-12-08): 0.2.6
- Removed `esp32FOTA::checkURL` and `esp32FOTA::useDeviceID`
- Improved memory management with config
- Added CI Tests for Arduino IDE
- Fixed `withDeviceID.ino` example

fixes #105 
fixes #115 
fixes #116
closed 2022-12-08 by tobozo