chrisjoyce911/esp32FOTA #118: No manifest_url provided in config, aborting!

milkywade, 2022-12-09:
Hi. I'm using 0.2.6.

I am running the HTTP example and when I execute the following call in the loop function:

`bool updatedNeeded = esp32FOTA.execHTTPcheck();`

...I receive the message error in the issue title.

There is well a manifest_url variable:

`const char *manifest_url = "http://192.168.1.100/ota/fota.json";`

...and this variable is the one referenced by the following call in the setup() function:

`esp32FOTA.setManifestURL( manifest_url );`

Any ideas ?

Thanks !

apppie123, 2022-12-10:
Yep, same issue here.

It seems that the last release of 2 days ago has broken this library.

I also see that a lot has changed in the HTTPS examples. They're suddenly all the same..

tobozo, 2022-12-10:
sorry about that, I pushed this release a bit too fast, working on a fix now

PR #119 (2022-12-10): 0.2.7
- Fixed config allocator
- Fixed examples
- Added `esp32FOTA::printConfig()`

fixes #118 (for real? :crossed_fingers:)
closed 2022-12-10 by tobozo