chrisjoyce911/esp32FOTA #121: Update failed

milkywade, 2022-12-11:
Disclaimer: this is the first time I get this far so I might be doing something wrong.

Issue: using version 0.2.7 when executing `esp32FOTA.execOTA();` (updatedNeeded is True) I get the following message:

```
Opening item http://192.168.1.100/ota/test_1-version-1_0_1.bin
This server supports resume!
Begin Firmware OTA. This may take 2 - 5 mins to complete. Things might be quiet for a while.. Patience!

abort() was called at PC 0x40082196 on core 1

Backtrace:0x40083881:0x3ffb23b00x4008e421:0x3ffb23d0 0x400935d9:0x3ffb23f0 0x40082196:0x3ffb2470 0x400ded2f:0x3ffb24c0 0x400d8111:0x3ffb24e0 0x4015386e:0x3ffb2500 0x40153afa:0x3ffb2520 0x400d5981:0x3ffb2540 0x400d626b:0x3ffb27b0 0x400d2be6:0x3ffb27f0 0x400da4a5:0x3ffb2820

ELF file SHA256: 0000000000000000

```
Rebooting...

I am using an ESP-CAM and have done no previous partition work or anything else than compiling & uploading binaries using arduino/platformIO.

Is the above command supposed to work in my case ?

If not, could I be pointed out to a resource describing the necessary preparatory work on the ESP, step by step ?

Many thanks!

milkywade, 2022-12-11:
FYI the file used as .bin is firmware.bin found on "C:\PlatformIO\Projects\Espressif ESP32 Dev Module\.pio\build\esp32cam"

milkywade, 2022-12-15:
This is the partition scheme being used: [https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv)

P.S. Using platform.io on VSCode so particion scheme is in a different place.

Humancell, 2022-12-15:
Yep.  Ok.  I was just curious if it could be the partition wasn't set-up for OTA.  And you can hit the URL "http://192.168.1.100/ota/test_1-version-1_0_1.bin" and download the file in your browser?

What options did you set for the config?

P.S. I'm just asking as I got it working on my ESP32 devices and it just worked.

milkywade, 2022-12-16:
Hey, thanks. Yes, the URL hits OK using the browser and the file downloads.

Re config, this is what I have:

>> printConfig()
```
Name: test_1
Manifest URL:http://192.168.1.100/ota/fota.json
Semantic Version: 1.0.1
Check Sig: false
Unsafe: false
Use Device ID: false
RootCA: false
PubKey: false
```

>> fota.json (note I manually change in this file the version to 1.0.2 whenever I want to test the OTA)
```
[
  {
    "type": "test_1",
    "version": "1.0.1",
    "url": "http://192.168.1.100/ota/test_1-version-1_0_2.bin"
  }
]
```

Last, the board is a espcam.

milkywade, 2022-12-17:
Just compiled using a different IDE: arduino v1.8.19.
Uploaded a new version to the ESPCAM, compiled again and uploaded "OTA_test_3.ino.bin" to the HTTP server (renamed as test_1-version-1_0_3.bin).

When adjusting the fota.json file to force the OTA process, the ESPCAM crashes with error:

```
Opening item http://192.168.1.100/ota/test_1-version-1_0_3.bin
This server supports resume!
Begin Firmware OTA. This may take 2 - 5 mins to complete. Things might be quiet for a while.. Patience!

abort() was called at PC 0x40082042 on core 1

Backtrace:0x40083709:0x3ffb23b00x4008c95d:0x3ffb23d0 0x40091a81:0x3ffb23f0 0x40082042:0x3ffb2470 0x400e8043:0x3ffb24c0 0x400da479:0x3ffb24e0 0x400d9de2:0x3ffb2500 0x400da02e:0x3ffb2520 0x400d5361:0x3ffb2540 0x400d592f:0x3ffb27b0 0x400d2c2e:0x3ffb27f0 0x400dbb45:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...

```

What am I doing wrong ?
Am I uploading the wrong BIN file ?

milkywade, 2022-12-17:
OK. I found it.

As explained here: [https://www.esp32.com/viewtopic.php?t=14700](https://www.esp32.com/viewtopic.php?t=14700)
boards.txt describes the esp32-cam as needing a huge_app partition.
`
esp32cam.build.partitions=huge_app`

When I change it to default, OTA works.

`esp32cam.build.partitions=default`

My eyes watered a little seeing the first OTA & the version numbers change :-)

Thank you @Humancell for putting me on the track !

closed 2022-12-17 by milkywade

Humancell, 2022-12-19:
This is awesome news!  It really is amazing to see it work!  I travel back and forth between the east and west coasts of the US and I love seeing my test devices reboot and report in with the new firmware version when I'm on the other side of the country!