chrisjoyce911/esp32FOTA #122: Feature request not bug for fota with flash encryption

simogaspa84, 2022-12-16:
Hi @chrisjoyce911 .. 

I would like to know if you have ever thought to test the esp32fota lib in case of the encryption of the flash.
There is a big and long topic regarding this 

https://github.com/espressif/arduino-esp32/issues/5645#issuecomment-961191953

Basically if you modify the bootloader creating a custom one.. you cna then encrypt the flash ..

I would like to know in your opinion if your lib could work also if the flash is encrypted how is shown in the topic above.

I hope to be enough clear otherwise tell me. 

Thanks

LeTuanAnhEP0818E, 2023-11-10:
I have the same question as you. What were the results? Hope the author can answer this question.

tobozo, 2024-08-19:
please see issue #153

closed 2024-08-19 by tobozo