chrisjoyce911/esp32FOTA #123: Cannot call setProgressCb

grunnels, 2022-12-30:
Maybe I'm missing something, but cannot call setProgressCb and I don't see it in the source code.  Is it in an upcoming release?

tobozo, 2022-12-30:
it's a callback setter, it accepts either a lambda function:

```cpp
void setup()
{
  // (...)
  FOTA.setProgressCb( [](size_t progress, size_t size) {
      if( progress == size || progress == 0 ) Serial.println();
      Serial.print(".");
  });
}
```

... or an existing function

```cpp
void my_progress_cb(size_t progress, size_t size) 
{
  if( progress == size || progress == 0 ) Serial.println();
  Serial.print(".");
}

void setup()
{
  // (...)
  FOTA.setProgressCb( my_progress_cb );
}

```

grunnels, 2022-12-30:
Thanks for the quick reply.  I'm sorry, it was a version mismatch error.  Great library, thanks for the work.

closed 2022-12-30 by grunnels