chrisjoyce911/esp32FOTA #136: Sharing an of my FOTA implementation.

hoeken, 2023-11-01:
This isn't really an issue, but I wanted to share my setup since I haven't really found a complete guide to using this library and thought it might be useful to others.

I'm using esp32FOTA in my open hardware project Yarrboard and find it very useful.  You can find the source here: https://github.com/hoeken/yarrboard

The setup is fairly straightforward, with signed binaries that are stored on github, along with the firmware.json file.  You can see how I use FOTA in this file: https://github.com/hoeken/yarrboard/blob/main/firmware/src/ota.cpp

I like to keep the keys stored in the binary as constants so that I don't have to deal with separately updating spiffs, and if I need to update keys at a later date, it will be handled seamlessly.  The spiffs update is nice, but it doesn't have the failsafe setup with 2 partitions like the main firmware.  If the spiffs update goes wrong, and you rely on spiffs, you could brick the unit.  No bueno.

The main upgrade that I wrote is a python script that will generate a new release: https://github.com/hoeken/yarrboard/blob/main/firmware/make_release.py

It is all based on platformio + vscode, and you tell it what environments to build binaries for.  I have an environment setup for each board that I am supporting, so it will build each different binary, sign it, put it in a releases folder, and then generate the firmware.json file all in one handy script.  I also add a "changelog" field to the firmware.json so that my web UI code can show the user what has changed in the new firmware.

Anyway, this project is great and I really appreciate the work the devs have put into it.

PS. this is only sort of related, but I found a really nice workaround for having web content, but not serving it from SPIFFS.  Basically I use gulp to take all the html + js + css + images and compile it into a single file that gets compiled into a single header file that you can #include and then access as a string.  This has the advantage of a single firmware.bin you can send OTA which contains both the code and the web UI for that version.  It also has the bonus of being cache friendly since you can use the __TIME__ and __DATE__ defines to send a modified date to the client.  I switched to the small spiffs partition setup, so I have more room for code size too.  I'm loving it so much more than serving all the static files from SPIFFS.  Gulp file here: https://github.com/hoeken/yarrboard/blob/main/firmware/gulpfile.js

Sorry about the wall of text, this post kinda got away from me.

TheDutchDev, 2024-01-01:
@hoeken links are broken since you've split up the project probably after this post?

for example, ota.cpp is now located here: https://github.com/hoeken/yarrboard-firmware/blob/main/src/ota.cpp

hoeken, 2024-01-01:
Yes sorry that is the correct link. Project got too big for a single repo.