chrisjoyce911/esp32FOTA #145: Stack overflow error.

astorun, 2024-04-22:
Thank you for opening an issue on an esp32OTA Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **Custom esp32-s3 board**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **ESP-IDF (6.5.0) and Arduino IDE in PlatformIO**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

Hi all,

I am getting an error with our esp32 FOTA code.

Here is portion of the code that we are using below.
We are calling in another part of the program.

The problem related with stack overflow does happens randomly. Here is the information how we call the code in background.

Board boots without any issue.
After 20 seconds, it will automatically connects to wifi.
After 60 seconds it will start checking if there is a firmware update and only if wifi connection is established.
Everything works fine here.

If we request another firmware update as soon as the initial firmware check it works normal.

However, after around a few minutes, if we request another software update check via user interface, everytime it results the esp32 halt just before the `[871411][V][ssl_client.cpp:269] start_ssl_client(): Performing the SSL/TLS handshake...`

It seems like it is happening only after TCP connection is lost.

Not sure how to fix this issue.

```
/**
   esp32 firmware OTA

   Purpose: Perform an OTA update to both firmware and filesystem from binaries located
            on a webserver (HTTPS) while using progmem to check for certificate validity
*/

#include "FS.h"
[... 176 lines cut ...]
[854401][V][ssl_client.cpp:321] stop_ssl_socket(): Cleaning SSL connection.
[854405][D][HTTPClient.cpp:642] sendRequest(): sendRequest code=-5

[854411][W][HTTPClient.cpp:1483] returnError(): error(-5): connection lost
[854417][D][esp32fota.cpp:831] execHTTPcheck(): Unknown HTTP response
[854424][D][HTTPClient.cpp:408] disconnect(): tcp is closed

[870936][I][esp32fota.cpp:815] execHTTPcheck(): Getting HTTP: redacted/fwDesc.json
[870936][I][esp32fota.cpp:360] setupHTTP(): Connecting to: redacted/fwDesc.json
[870947][V][HTTPClient.cpp:252] beginInternal(): url: redacted/fwDesc.json
[870957][D][HTTPClient.cpp:303] beginInternal(): protocol: https, host: redacted port: 443 url: redacted/fwDesc.json
[870969][D][HTTPClient.cpp:598] sendRequest(): request type: 'GET' redirCount: 0

[871350][V][ssl_client.cpp:62] start_ssl_client(): Free internal heap before TLS 122927
[871350][V][ssl_client.cpp:68] start_ssl_client(): Starting socket
[871390][V][ssl_client.cpp:146] start_ssl_client(): Seeding the random number generator
[871393][V][ssl_client.cpp:155] start_ssl_client(): Setting up the SSL/TLS structure...
[871395][D][ssl_client.cpp:176] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[871403][V][ssl_client.cpp:254] start_ssl_client(): Setting hostname for TLS session...
[871411][V][ssl_client.cpp:269] start_ssl_client(): Performing the SSL/TLS handshake...

***ERROR*** A stack overflow in task esp_timer has been detected.

Backtrace: 0x40377981:0x3fcf2b00 0x4037e899:0x3fcf2b20 0x40381e62:0x3fcf2b40 0x40380881:0x3fcf2bb0 0x4037e950:0x3fcf2bd0 0x4037e946:0x3fcf2be0 0x733ffffc:0x420a8633 |<-CORRUPTED
```

tobozo, 2024-04-23:
hi and thanks for your feedback 👍 

first call

    Free internal heap before TLS 123043
    ....
    Free internal heap after TLS 91859

second call

    Free internal heap before TLS 122927 <<< 116 bytes lost
    !! stack overflow when Verifying peer X.509 certificate... !!
    
116 bytes mem leak doesn't seem much; but it's still a strong sign of memory fragmentation

the issue can probably be solved by freeing around 10KB or heap or by using an esp32 with psram

there's not much esp32FOTA library can do since it happens while x509 is being verified by the espressif core, maybe try it with an older core version?

astorun, 2024-04-23:
Thanks for pointing that out!

We actually realized SPI-RAM was not properly initialized for our ESP32-S3 and fixed it.
Diverted some resources to external memory to open space in internal memory.

Now we have a ton of free internal memory.

However the issue still persists even there is a more than 180kb free internal memory available.
It happens only after tcp connection is lost and trying to connect again on the third tryout.

```
12:44:51:102 -> =======================================================
12:44:51:102 -> 
12:44:51:102 -> Machine Name: 
12:44:51:108 -> GR-PRO-R3
12:44:51:108 -> Machine Version: 
12:44:51:108 -> GR PRO R3
12:44:51:108 -> Firmware version: 9.9.9-DEV
12:44:51:113 -> 
12:44:51:113 -> MCU: ESP32-S3 Rev 0
12:44:51:113 -> This chip has 2 cores
12:44:51:113 -> Board ID: 48:27:E2:8F:2A:D8
12:44:51:119 -> 
12:44:51:119 -> CPU Freq: 240 MHz
12:44:51:119 -> XTAL Freq: 40 MHz
12:44:51:126 -> APB Freq: 80000000 Hz
12:44:51:126 -> Free heap memory: 296063 bytes
12:44:51:132 -> Free PSRAM: 2094851 bytes
12:44:52:479 -> [  1798][D][esp32fota.cpp:140] setString(): Assigned value: redacted <= redacted 
12:44:52:485 -> [  1798][D][esp32fota.cpp:140] setString(): Assigned value: redacted  <= redacted 
12:44:52:515 -> Free PSRAM: 1898043 bytes
12:45:07:470 -> [ 16789][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 0 - WIFI_READY
12:45:07:507 -> [ 16826][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:45:07:513 -> [ 16827][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
12:45:07:518 -> [ 16832][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 2 - STA_START
12:45:11:922 -> [ 21240][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: redacted , BSSID: redacted , Reason: 2
12:45:11:927 -> [ 21242][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:45:11:938 -> [ 21248][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
12:45:11:944 -> [ 21255][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running
12:45:11:949 -> [ 21263][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:45:16:342 -> [ 25660][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: ASTRN, BSSID: 80:cc:9c:3e:d6:04, Reason: 2
12:45:16:347 -> [ 25662][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:45:16:358 -> [ 25668][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
12:45:16:364 -> [ 25675][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
12:45:16:369 -> [ 25684][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:45:20:776 -> [ 30095][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: redacted , BSSID: redacted , Reason: 2
12:45:20:782 -> [ 30097][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:45:20:793 -> [ 30103][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
12:45:20:798 -> [ 30110][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
12:45:20:804 -> [ 30118][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:45:22:456 -> [ 31775][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:45:22:461 -> [ 31776][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: ASTRN, BSSID: 80:cc:9c:3e:d6:04, Reason: 202
12:45:22:792 -> [ 32106][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:45:22:797 -> [ 32109][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 202 - AUTH_FAIL
12:45:22:803 -> [ 32118][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: ASTRN, BSSID: 80:cc:9c:3e:d6:04, Reason: 202
12:45:22:814 -> [ 32126][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:45:22:825 -> [ 32134][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 202 - AUTH_FAIL
12:45:25:229 -> [ 34548][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: ASTRN, BSSID: 00:00:00:00:00:00, Reason: 205
12:45:25:234 -> [ 34549][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:45:25:246 -> [ 34556][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 205 - CONNECTION_FAIL
12:45:25:251 -> [ 34563][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
12:45:25:256 -> [ 34572][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:45:29:723 -> [ 39041][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: ASTRN, BSSID: 80:cc:9c:3e:d6:04, Reason: 2
12:45:29:729 -> [ 39044][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
12:45:29:740 -> [ 39050][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
12:45:29:745 -> [ 39057][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
12:45:29:751 -> [ 39065][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
12:45:30:676 -> [ 39996][V][WiFiGeneric.cpp:355] _arduino_event_cb(): STA Connected: SSID: ASTRN, BSSID: 80:cc:9c:3e:d6:04, Channel: 7, Auth: WPA2_WPA3_PSK
[... cut ...]
13:32:49:821 -> SPIWP:0xee
13:32:49:827 -> mode:DIO, clock div:1
13:32:49:827 -> load:0x3fce3808,len:0x470
13:32:49:827 -> load:0x403c9700,len:0xc3c
13:32:49:832 -> load:0x403cc700,len:0x2e80
13:32:49:832 -> entry 0x403c98e8
13:32:50:186 -> [   374][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
13:32:50:220 -> [   410][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=2 scl=1 freq=400000
13:32:50:231 -> 
13:32:50:231 -> =======================================================
```

tobozo, 2024-04-23:
> ESP-IDF (6.5.0) 

have you tried idf menuconfig and extend the value of CONFIG_ESP_TIMER_TASK_STACK_SIZE ?

astorun, 2024-04-23:
I tried to increase it before fixing the issue with psram but with no success a day ago.

However, now per your suggestion, I increased it again from the default value and it appears it is working for now.

I did also some additional tweaks, which I suspect that might have some effect to other numerous settings in menuconfig.

```
CONFIG_ESP_TIMER_TASK_STACK_SIZE=8192

CONFIG_ARDUINO_LOOP_STACK_SIZE=16384
CONFIG_ASIO_SSL_SUPPORT=y
CONFIG_ASIO_USE_ESP_OPENSSL=y
CONFIG_ESP_TLS_CLIENT_SESSION_TICKETS=y
CONFIG_ESP_TLS_PSK_VERIFICATION=y
CONFIG_ESP_TLS_INSECURE=y
CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=y
CONFIG_ESP_HTTP_CLIENT_ENABLE_BASIC_AUTH=y

CONFIG_MBEDTLS_PSK_MODES=y
CONFIG_MBEDTLS_KEY_EXCHANGE_PSK=y
CONFIG_MBEDTLS_KEY_EXCHANGE_DHE_PSK=y
CONFIG_MBEDTLS_KEY_EXCHANGE_ECDHE_PSK=y
CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_PSK=y

CONFIG_TIMER_TASK_STACK_SIZE=8192
```

I will test it further and report back if the issue persists.

closed 2024-04-25 by astorun

d4rkdpg, 2024-04-25:
Hi m having simillar issues: abort() was called at PC 0x40376bfa on core .  

`This server supports resume!
Begin Firmware OTA. This may take 2 - 5 mins to complete. Things might be quiet for a while.. Patience!

abort() was called at PC 0x40376bfa on core 1

Backtrace: 0x40378032:0x3fceb9d0 0x4037df55:0x3fceb9f0 0x403843f5:0x3fceba10 0x40376bfa:0x3fceba90 0x4204fea3:0x3fcebae0 0x42045565:0x3fcebb00 0x420c8576:0x3fcebb20 0x420c87e3:0x3fcebb40 0x42042299:0x3fcebb60 0x4204292b:0x3fcebde0 0x4200b035:0x3fcebe30 0x4200b08f:0x3fcebeb0 0x42048909:0x3fcebf40` 

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420e2682
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
[   303][D][esp32FOTA.cpp:141] setString(): Assigned value: esp32-fota-http <= esp32-fota-http
[   303][I][esp32FOTA.cpp:901] debugSemVer(): Current firmware version: 1.0.6
[   324][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled

 Any pointers would be appreciated.