chrisjoyce911/esp32FOTA #146: Library working for esp32-c6?

simogaspa84, 2024-05-13:
Hi @chrisjoyce911 

Just want to ask if you see any issue for making this library work for esp32-c6 ?

Thanks a lot

tobozo, 2024-05-13:
hi,

esp32-c6 requires arduino ide 2.x.x (beta) and arduino-esp32 core 3.x.x (rc), none of which are in a stable state

let's keep this issue open until both of these packages produce a stable version?

tobozo, 2024-05-28:
hi @simogaspa84 the new [branch 0.2.8](https://github.com/chrisjoyce911/esp32FOTA/tree/0.2.8) of esp32FOTA library supports [arduino-esp32 core 3.0.0](https://github.com/espressif/arduino-esp32/releases/tag/3.0.0).

it compiles with esp32-c6 board without errors but I don't have a C6 to test with, can you give it a try?

commit 23489cad (2024-06-15): arduino-esp32 core 2.x => 3.x migration (fixes #146)
closed 2024-06-15 by tobozo