chrisjoyce911/esp32FOTA #150: mbedtls compile errors

aeonSolutions, 2024-06-14:
hi there,
i've installed esp32fota with the lastest version of espressIF libraries for arduino. 
when trying to compile using arduino studio 2.x I get the following error codes:

In file included from C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/build_info.h:144,
                 from C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/net_sockets.h:30,
                 from C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\NetworkClientSecure\src/ssl_client.h:8,
                 from C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\NetworkClientSecure\src/NetworkClientSecure.h:26,
                 from C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\NetworkClientSecure\src/WiFiClientSecure.h:2,

C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/check_config.h:694:2: error: #error "MBEDTLS_PLATFORM_STD_EXIT defined, but not all prerequisites"
  694 | #error "MBEDTLS_PLATFORM_STD_EXIT defined, but not all prerequisites"
      |  ^~~~~
C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/check_config.h:705:2: error: #error "MBEDTLS_PLATFORM_STD_FPRINTF defined, but not all prerequisites"
  705 | #error "MBEDTLS_PLATFORM_STD_FPRINTF defined, but not all prerequisites"
      |  ^~~~~
C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/check_config.h:710:2: error: #error "MBEDTLS_PLATFORM_STD_PRINTF defined, but not all prerequisites"
  710 | #error "MBEDTLS_PLATFORM_STD_PRINTF defined, but not all prerequisites"
      |  ^~~~~
C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/check_config.h:715:2: error: #error "MBEDTLS_PLATFORM_STD_SNPRINTF defined, but not all prerequisites"
  715 | #error "MBEDTLS_PLATFORM_STD_SNPRINTF defined, but not all prerequisites"
      |  ^~~~~
exit status 1

Compilation error: exit status 1

any assistance is greatly appreciated. 
thank you

tobozo, 2024-06-14:
hi,

does the branch [0.2.8](https://github.com/chrisjoyce911/esp32FOTA/tree/0.2.8) of this repo fix these errors?

aeonSolutions, 2024-06-14:
let me try..... brb

aeonSolutions, 2024-06-14:
just finished compiling. Same errors .

tobozo, 2024-06-15:
last build (5 days ago) worked fine with Arduino CLI + arduino-esp32 core v3.0.1

https://github.com/chrisjoyce911/esp32FOTA/actions/runs/9267513632/job/26032115767

also tried the anyFS.ino example on my system and can't reproduce the error

I'm not sure what "Arduino Studio" is (google says it's an Android app?) but from the logs you pasted it seems to be "Arduino IDE" with arduino-esp32 core v3.0.1.

I need more info in order to reproduce the issue, please provide a minimal sketch. If the code is more that 50 lines create a [gist](https://gist.new).

aeonSolutions, 2024-06-15:
here's a snapshot of the Arduino IDE with the errors codes
[image]

unfortunately, the number of lines of code is huge and this means I cannot put it on a gist here....

tobozo, 2024-06-15:
unfortunately without a way to reproduce the issue I can only assume it's caused by something outside the scope of the library

so either you provide such minimal example and I can use my free time to reproduce and fix it, or I'll close this issue as out of scope

aeonSolutions, 2024-06-15:
ok. Let me try with one of the ESP32FOTA examples ...brb

aeonSolutions, 2024-06-15:
Ok. here it is. I've tried compiling the example HTTP.ino . there's the output errors 

[image]

In file included from C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/build_info.h:144,
                 from C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/net_sockets.h:30,
                 from C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\NetworkClientSecure\src/ssl_client.h:8,
                 from C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.1\libraries\NetworkClientSecure\src/NetworkClientSecure.h:26,
                 from c:\Users\mtpsi\OneDrive\Documents\Arduino\libraries\esp32FOTA\src/esp32FOTA.hpp:45,
                 from c:\Users\mtpsi\OneDrive\Documents\Arduino\libraries\esp32FOTA\src/esp32fota.h:8,
                 from C:\Users\mtpsi\AppData\Local\Temp\.arduinoIDE-unsaved2024515-6944-mk24ie.ljpy\HTTP\HTTP.ino:17:
C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/check_config.h:694:2: error: #error "MBEDTLS_PLATFORM_STD_EXIT defined, but not all prerequisites"
  694 | #error "MBEDTLS_PLATFORM_STD_EXIT defined, but not all prerequisites"
      |  ^~~~~
C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/check_config.h:705:2: error: #error "MBEDTLS_PLATFORM_STD_FPRINTF defined, but not all prerequisites"
  705 | #error "MBEDTLS_PLATFORM_STD_FPRINTF defined, but not all prerequisites"
      |  ^~~~~
C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/check_config.h:710:2: error: #error "MBEDTLS_PLATFORM_STD_PRINTF defined, but not all prerequisites"
  710 | #error "MBEDTLS_PLATFORM_STD_PRINTF defined, but not all prerequisites"
      |  ^~~~~
C:\Users\mtpsi\AppData\Local\Arduino15\packages\esp32\tools\esp32-arduino-libs\idf-release_v5.1-442a798083\esp32c3/include/mbedtls/mbedtls/include/mbedtls/check_config.h:715:2: error: #error "MBEDTLS_PLATFORM_STD_SNPRINTF defined, but not all prerequisites"
  715 | #error "MBEDTLS_PLATFORM_STD_SNPRINTF defined, but not all prerequisites"

tobozo, 2024-06-15:
works for me

[image]

it looks like your IDE is still pulling the old version of esp32FOTA

I've pushed some changes on the [0.2.8 branch](https://github.com/chrisjoyce911/esp32FOTA/tree/0.2.8) so that the library.json and library.properties files have a higher version number

can you please delete esp32FOTA library from your libraries folder, and reinstall [esp32FOTA 0.2.8](https://github.com/chrisjoyce911/esp32FOTA/archive/refs/heads/0.2.8.zip), 

if your IDE had a duplicate installation of the library it should now pick the most recent version.

... then compile and look at the top of the logs for such an entry:

Alternatives for esp32fota.h: [esp32FOTA@0.2.8]
ResolveLibrary(esp32fota.h)
  -> candidates: [esp32FOTA@0.2.8]

aeonSolutions, 2024-06-15:
ok. let me try. brb

aeonSolutions, 2024-06-15:
All working now on my end too !
keep up the aweome work .

btw, i have coded my own libraries for arduino. If anyone is of interest to you ....
[https://github.com/aeonSolutions/aeonlabs-open-software-catalogue](https://github.com/aeonSolutions/aeonlabs-open-software-catalogue)

[image]

closed 2024-06-15 by aeonSolutions

tobozo, 2024-06-15:
thanks for your feedback, I'll publish a new release soon