chrisjoyce911/esp32FOTA #152: Ethernet usage

infrafast, 2024-06-29:
Hello this library looks really good! My use case is that my esp device can connect the network via wifi OR ethernet, so my question is would that still works over ethernet (w5500 using ethernet.h) thanks!

tobozo, 2024-06-29:
hi,

esp32FOTA uses the new [NetworkClient](https://github.com/chrisjoyce911/esp32FOTA/blob/master/src/esp32FOTA.hpp#L45) from espressif when applicable (needs arduino-esp32 3.x.x), so as long as your ethernet library uses it and Network.begin() has been called, it should work

infrafast, 2024-07-02:
Very good thanks for your answer, I am using ethernet.h and therefore call Ethernet.begin() or WiFi.begin() depending if I detect eth HW  or not but I will have a look to Network.begin as you mention and see if I can migrate

infrafast, 2024-07-10:
Hello, I gave it a try but can't compile. If I just add "chrisjoyce911/esp32FOTA" to my libs in platformIO,  I get a full bunch of erros (see below) 

my env is the following:

[env]
platform = espressif32@6.4.0                                    ; 3.20011.230801 (2.0.11)
board = seeed_xiao_esp32s3
framework = arduino

[{
	"resource": "/c:/PlatformIO/esp32/.pio/libdeps/dev/esp32FOTA/src/esp32FOTA.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'_http' was not declared in this scope",
	"startLineNumber": 366,
	"startColumn": 5,
	"endLineNumber": 366,
	"endColumn": 5
},{
	"resource": "/c:/PlatformIO/esp32/.pio/libdeps/dev/esp32FOTA/src/esp32FOTA.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'HTTPC_STRICT_FOLLOW_REDIRECTS' was not declared in this scope",
	"startLineNumber": 366,
	"startColumn": 30,
	"endLineNumber": 366,
	"endColumn": 30
},{
	"resource": "/c:/PlatformIO/esp32/.pio/libdeps/dev/esp32FOTA/src/esp32FOTA.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "'_http' was not declared in this scope",
	"startLineNumber": 449,
	"startColumn": 13,
	"endLineNumber": 449,
	"endColumn": 13
},

etc...

could you help?
Thank you

tobozo, 2024-08-19:
hi,

the official platformio package doesn't seem to be maintained any more, please try with this platform version instead

```ini
[env]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
```

infrafast, 2024-09-28:
Thank you Tobozo for the advise, I've tested but unfortunately it is not compatible either, generate a lot of compilaiton errors... I will further check if I can make this working and feedback here but it will take some time

tobozo, 2024-09-28:
Sometimes platformio compilation errors can be caused (or be fixed by) tweaking `lib_ldf_mode`.
But more important is how esp32FOTA is included from your project. The fool-proof approach is to include every library in `lib_deps` from your main.cpp only.

If possible please share a platformio minimal project that reproduces these errors, I may be more helpful with that.