chrisjoyce911/esp32FOTA #157: Use custom networking stack?

anthonyhoegberg, 2024-09-12:
So i have a question if its possible to instead of using networkclient built in to esp32 if we can use custom networking like
https://github.com/govorox/SSLClient
Or other Client based transports.

That way we get support for many different Wifi, Sim and Ethernet modems that implements a client class. And ssl on top of that implemented by SSLClient using those clients etc.

tobozo, 2024-09-12:
hi, thanks for your feedback :+1: 

SSLClient and NetworkClientSecure inherit from the same Arduino Client class and share the client method names used by esp32FOTA, so it should probably work.

you can experiment by modifying the include block in [esp32FOTA.hpp](https://github.com/chrisjoyce911/esp32FOTA/blob/master/src/esp32FOTA.hpp#L43C1-L50).

```cpp

#if __has_include(<SSLClient.h>)
  #include <SSLClient.h> // https://github.com/govorox/SSLClient/
  #define ClientSecure SSLClient
#elif __has_include(<NetworkClientSecure.h>)
  #include <NetworkClientSecure.h> // arduino-esp32 core 3.x
  #define ClientSecure NetworkClientSecure
#else
  #include <WiFiClientSecure.h> // // arduino-esp32 core 2.x
  #define ClientSecure WiFiClientSecure
#endif

```

if this works for you, feel free to submit a pull request :wink:

anthonyhoegberg, 2024-09-17:
the specific instance i provide needs to be configured before like sim settings etc. before i can leave control over to the update library, so just replacing the type wouldnt work? i would need to pass the network instance to be used to the library i think?

tobozo, 2024-09-28:
been trying to get esp32FOTA to work with phy/transport/app layers without refactoring every OSI related function calls but failed so I guess a refactor will be needed after all

I'm testing a template that I hope will accept both custom and builtin phy/transport/app clients:

```cpp
template<typename APP, typename PRES, typename PHY>
struct OSILayers
{
  OSILayers() { };
  OSILayers( APP& app, PRES& pres, PHY& phy )
    : AppClient(&app), PresClient(&pres), PhyClient(&phy) { }
  APP  AppClient;   // application (HTTP, Pubsub)
  PRES PresClient;  // Presentation (SSL, Network)
  PHY  PhyClient;   // Transport/Phy (WiFi, Ethernet, GSM)
};

HTTPClient appclient;
NetworkClientSecure presclient;
NetworkClient phyclient;
auto builtinStack = new OSILayers<HTTPClient*,NetworkClientSecure*,NetworkClient*>(&appclient, &presclient, &phyclient);

// PubSubCLient appclient;
// SSLClient presclient;
// TinyGSMClient phyclient;
// auto customStack = OSILayers<PubSubCLient*,SSLClient*,TinyGSMClient*> stackFOTA(&appclient, &presclient, &phyclient);
```

This `OSILayers` isn't connected to anything yet, but it is intended to be used from within the esp32FOTA class.
Any existing code specific to the esp32 builtin network stack will have to move out of esp32FOTA class.

This creates a significant tech debt:

1) Need to reimplement all OSI related functions (e.g. add auth headers to http) as callbacks
2) Provide a default `OSILayers` instance based on arduino-esp32 built-in stacks (v2.x.x and v3.x.x) to avoid breaking existing projects, which should be ignored when a custom stack is used
3) Create some `OSILayers` examples based on custom stacks
4) Unit tests :grimacing:

anthonyhoegberg, 2024-09-30:
Sounds like its quite complicated to get this working if I'm correct?

Humancell, 2024-12-01:
Hello,

I use your wonderful OTA code for all of my WiFi projects ... it's great!

I also am using TinyGSM and a cellular modem for connectivity from some Lilygo boards.  Currently, to provide HTTP for my application I use code similar to this:

```
  // HTTP is using modem connection 0
  #if USE_HTTPS
    TinyGsmClientSecure http_client(modem,0);
    HttpClient http(http_client, server, port);
  #else
    TinyGsmClient http_client(modem,0);
    HttpClient http(http_client, server, port);
  #endif
#endif
```
Would there be a way that I could pass a reference to this HttpClient into the code to allow this HTTP client to be used for the OTA transactions?  I just started to look through your code to see what might be possible, but don't fully understand it yet.

Thank you either way for creating such a great library!